micro-nova/AmpliPi #54: Add openHAB Integration/bindings

linknum23, 2021-01-28:
TBD

kaikreuzer, 2021-07-12:
Work has started - please see https://github.com/openhab/openhab-addons/pull/10983.

kaikreuzer, 2021-07-14:
Hey @linknum23, after having started with the binding implementation and modelling the AmpliPi itself as a bridge with a channel to choose its presets, I am wondering how I should now best model the "sub-things".
In your web app, you have the 4 sources as the leading concept and let the user dynamically assign zones and groups to them.
From a home automation perspective, I would instead see the locations (i.e. zones and groups) be the leading concept and let the user choose which source to play there and at what volume.
I would therefore tend to implement an auto-discovery, which would add a Thing for each zone and each group that is found. Those Things would then have a "source", "volume" and "mute" channel each.
The options for the source channel would be a combination of the sources and streams, i.e. "RCA1", "RCA2", "RCA3", "RCA3", "shairport", "dlna", "Beatles Radio", etc. The only limitations here would be that the set of selected sources of all zones+groups cannot exceed 4 - not sure yet, how I'd go about that.

Does this approach make sense to you? Is there anything that I might oversee that is important to consider?

linknum23, 2021-07-14:
I see where you are coming from and from what I understand I think that approach makes sense. The pain point I see you having is dealing with determining which source# to use. 

Right now to handle the equivalent of the voice command **Play "My favorite radio station" on the "Patio" at 50% volume** a client interface needs to execute the following API requests:

1. `get_status` : parse then find an available audio 'source' X (or the one already playing "My favorite radio station" call that X)
2. `set_source`: set source X's input to "My favorite radio station" (not necessary if it is already playing)
3. `set_zone`: set zone Patio's volume to 50% and its source to X

We may want to simplify that command sequence so that the client doesn't need all of the domain knowledge needed to find an available source. 

Are there other specific use cases/workflows you have in mind for openHAB? We can suggest some API sequences to fit different use cases, or adjust the API to make certain workflows easier to accomplish.

kaikreuzer, 2021-07-15:
I don't think the number of calls is an issue here.

1. The `get_status` will anyhow be continuously polled and cached locally, so openHAB is always aware of the state of the AmpliPi.
2. I'd assume the user can define his "3 favorite" sources, which could simply be kept and the 4th could be dynamically swapped between other, not that regularly used sources (assuming that those are not an RCA). Maybe I'd allow to "pin" certain sources and use the others in a FIFO way - will have to think about it.
3. Setting the zone to source x is the main call. If a source is playing, adjusting the volume is another use case, which again would only require a single call.

-> I think the current API structure is just fine and will do.

> Are there other specific use cases/workflows you have in mind for openHAB? 

What definitely is another use case that does a lot at once is the PA feature - but you know those requirements already and I am very much looking forward to your work in https://github.com/micro-nova/AmpliPi/pull/135.

The other feature that will make the openHAB integration smoother is the [SSE support](https://github.com/micro-nova/AmpliPi/issues/739) - at the moment I am polling at a fixed interval (default 10 seconds), which clearly is not as nice than when changes are immediately pushed to openHAB.

kaikreuzer, 2021-07-17:
@linknum23 One feature thatI forgot to mention and which would be worthwhile for a smoother integration is mDNS - I didn't check if this might be in place already, but didn't find it mentioned anywhere. If the AmpliPi announces itself through mDNS on the network, openHAB could automatically discover it and add it to its Inbox.

linknum23, 2021-07-19:
AmpliPi piggybacks off of the default debian/raspbian mDNS integration using the .local domain. We ship all of the main controllers with a hostname of amplipi, so that they can be accessed via mDNS using amplipi.local. Is that useful enough?

linknum23, 2021-07-19:
There has been a change to the API from #121 stream.info has been migrated to source.info. I hope this doesn't cause issues for this integration. The intent was to provide a simplified metadata interface that also includes source.info.name which has a friendly display name for the audio source like 'Beatles Radio - internet radio' or 'Input 1 - rca'.

kaikreuzer, 2021-07-19:
> We ship all of the main controllers with a hostname of amplipi, so that they can be accessed via mDNS using amplipi.local. Is that useful enough?

I'm not sure. This sounds to me like "DHCP Client Option 12", i.e. providing a hostname for the server.
For mDNS (aka Bonjour), a server rather announces the services it has to offer. I would expect the AmpliPi to announce a service type "_http._tcp", which then points to its REST API. Note that with mDNS, you can have multiple AmpliPis in your local network and all of them could be found (as their service broadcast includes their respective IP addresses).
Running a 
``` 
dns-sd -B _services._dns-sd._udp
```
from command line lists all servers that do some mDNS announcements in the network. I cannot find the AmpliPi in the result list.

> There has been a change to the API from #121 stream.info has been migrated to source.info.

Sounds reasonable, thanks for the heads-up. Will have to update my AmpliPi then, so that I develop against the correct version of the API.

linknum23, 2021-07-19:
It looks pretty easy to register a service for AmpliPi. I have a prototype working right now that advertises `/api`. We will try to add this soon.

kaikreuzer, 2021-07-20:
Sounds great, thanks!

kaikreuzer, 2021-07-26:
@linknum23 Stupid question: What's the simplest way to update my AmpliPi to the latest codebase, so that it includes #121? Shall I follow [these 4 steps](https://github.com/micro-nova/AmpliPi/blob/master/docs/DEVELOPING.md#developing-on-an-amplipi-controller-over-ssh)? If so, isn't there a step missing, which is shutting down the web server first?

linknum23, 2021-07-26:
Let me make this easy for you to start with. Here is the latest code on master: [amplipi-0.1.5alpha0.tar.gz](https://github.com/micro-nova/AmpliPi/files/6881225/amplipi-0.1.5a0.tar.gz)

To update from the app:

1. Download this update file
2. Click the gear icon next to source 4 on the amplipi app
3. Click `Check for updates`
4. Browse for the update downloaded in step 1
5. Click `Upload software`
6. Wait for the page to redirect back to the app, it should take 1-2 minutes

Otherwise, if you have a checkout of the latest master branch of amplipi on your computer:
1. Run `./scripts/deploy`
1. Wait a couple of minutes for it to copy and configure the latest code. The version reported should be 0.1.5.alpha0

kaikreuzer, 2021-07-27:
Perfect, thanks, that worked! 👍

kaikreuzer, 2021-08-02:
Good news everyone - the first version of the AmpliPi binding has been merged and is already available in openHAB 3.2 Milestone 1: https://next.openhab.org/addons/bindings/amplipi/

@linknum23 I guess this issue can hence be closed.
I will continue working on enhancements, especially once SSE and mDNS become available (I have created https://github.com/micro-nova/AmpliPi/issues/152 to track progress on mDNS as there wasn't any issue yet afaik).

closed 2021-08-02 by linknum23