micro-nova/AmpliPi #62: Add an option to install Roon Bridge

wjanssen, 2021-02-15:
[Roon](https://roonlabs.com/) is a widely used music player software which streams locally from a "core" on the LAN to audio devices that support network access.  It can be supported on audio equipment, include Pi-based devices like AmpliFi, by downloading and installing the Linux version of [Roon Bridge](https://help.roonlabs.com/portal/en/kb/articles/roonbridge).  It would be great if the AmpliFi web page had a button which would cause it to [download and install Roon Bridge](https://help.roonlabs.com/portal/en/kb/articles/linux-install).

linknum23, 2021-02-16:
So as far as I can tell AmpliPi could easily add at least one Roon bridge, potentially up to 4 since AmpliPi has 4 independent stereo channels. One possible complication is that a Roon bridge can switch between stereo channels independently of the AmpliPi unit which could cause conflicts ( such as multiple songs playing out the same speaker ).

I am interested how your setup would look like with both a Roon and an AmpliPi unit in the same house. Could you explain how you plan to use this in combination with a Roon controller?

I don't know that much about the Room system...

linknum23, 2021-02-16:
I looked into this a little further and it seems like the easiest way to integrate with Roon currently is to use AmpliPi's existing airplay interface. AmpliPi can play 4 different airplay streams at once, each connected to one or multiple zones.

linknum23, 2021-02-22:
Looking into this further, I don't think adding a Roon Bridge makes much sense initially, since the use case of the Roon Bridge is covered by the existing Airplay interface. 

Full Roon integration, at a "Roon Ready" level would likely involve porting the whole AmpliPi software system to Roon's JS framework. This would be outside the scope of our goals for software, but since the whole system is open it could be taken up by the community with our assistance. @wjanssen do you know of anyone from the Roon community be interested in adding that support?

linknum23, 2021-02-23:
Closing this for now, since we haven't really gotten any interest in this feature. Feel free to open it back up.

closed 2021-02-23 by linknum23

wjanssen, 2021-02-28:
Just to clarify: "Roon Ready" is a fairly heavyweight commitment which would include certification from Roon Labs.  It involves signing an NDA and getting the SDK from Roon, and hooking up the SDK correctly to your software, then sending a unit to Roon Labs for testing and certification.

"Roon Bridge", on the other hand, is a drop-in software component that doesn't require tinkering with your software.  It uses the standard audio frameworks for the OS you install it on (I presume this would be ALSA for your presumably Linux platform) to see what the audio capabilities are and manipulate them.

My suggestion was that you add a button which would cause your system to download and install Roon Bridge.  This would automatically make your system visible to a Roon Core running elsewhere on the LAN.  How Roon would "see" your system is an interesting question, one perhaps most easily answered simply by downloading Roon Bridge and installing it, to see what happens.  If it works, great!  If not, this would become a more labor-intensive sub-project.