micro-nova/AmpliPi #66: Add Mopidy and USB storage

bs42, 2021-02-19:
[Mopidy](https://mopidy.com/) is an open source music player that can play local files as well as some online sources and has a full API.

This combined with USB storage would enable the AmpliPi to play completely offline music on the system without the need for Internet or a second device (other than a USB Storage plugged into one of the USB ports).

linknum23, 2021-02-19:
It looks like it should be pretty easy to install:
```
sudo apt install mopidy
sudo ln -s ~/.asoundrc /etc/asound.conf
```
The configuration file, mopidy.conf, needs to be generated with something like this, where X is the source id and NAME is stream name:
```
[http] port = 6X88
[http] zeroconf = NAME on AmplPi
cache_dir = /home/pi/config/srcs/X/mopidy/cache
config_dir =  /home/pi/config/srcs/X/mopidy/config
data_dir =  /home/pi/config/srcs/X/mopidy/data
[local]
media_dir = /path/to/usb # figure out how to automount first USB device to same place?
[audio]
output = alsasink device=chX
```

Based on what I read this should support multiple instances although the above config will need some tweaking.

Further reading:
- https://docs.mopidy.com/en/latest/config/
- https://discourse.mopidy.com/t/config-and-initscripts-for-multiple-mopidy-servers/1017

bs42, 2023-01-09:
Just wanted to update this.  After serious procrastination and then removing the only device I had with audio-out and then realizing it was what was feeding my AmpliPi, I finally ended up implementing this and it is fantastic.  I have a 128gb usb stick plugged in for the music currently, though I may need to upgrade that for more space soon.

A couple notes for anyone else wanting to implement this as well as notes for if this is to be implemented official.

The debian release on the AmpliPi is too old for a current version of Mopidy so use their repo and update: [https://docs.mopidy.com/en/latest/installation/debian/#install-from-apt-mopidy-com](url)

Here's the config I ended up with:
```
[http] 
port = 6188
hostname = ::
zeroconf = Mopidy on AmpliPi

[core]
cache_dir = /home/pi/config/srcs/1/mopidy/cache
config_dir =  /home/pi/config/srcs/1/mopidy/config
data_dir =  /media/pi/Stuff/mopidy/data

[file]
enabled = false

[mpd]
enabled = true
hostname = ::

[local]
media_dir = /media/pi/Stuff
included_file_extensions = 
  .mp3 
  .flac 
  .wav

[m3u]
playlists_dir = /media/pi/Stuff/playlists
base_dir = /home/pi/config/srcs/1/mopidy/m3u

[audio]
output = alsasink device=ch0

[iris]
enabled = true
country = US
locale = en_US
spotify_authorization_url = http://localhost
lastfm_authorization_url = http://localhost
genius_authorization_url = http://localhost
data_dir = /home/pi/config/srcs/1/mopidy/iris
```

``/media/pi/Stuff/`` is my USB

The big thing to understand about this method is that the first Source on the AmpliPi (if using ch0 above) will be Mopidy regardless of what you have selected for a stream.  Running this on the AmpliPi basically takes over this interface.  If this is made official there will need to be a method in which the mopidy service can be kept running but not attached to the channel permanently.   I don't have enough knowledge of the hardware/software interface to pull that off on my own.

Personally I think this is a killer feature to implement in an official capacity, but I also understand I might be in the minority there and am really just super happy that this platform is open enough that I can do this on my device.

linknum23, 2023-01-10:
Thanks for sharing this setup! Something like this will be much more flexible once all of the details of #262 are figured out. At that point it won't use up a source when it isn't playing something.

wasik, 2023-06-10:
This seems like a great feature; officially supporting local audio in the future could a lot of value to the AmpliPi. Not having to manage a separate audio server would certainly simplify some home setups. Thanks for looking into it @bs42 !

linknum23, 2023-06-12:
One other option that is fairly easy to setup is installing Logitech media server on the amplipi.

juhaj, 2024-06-16:
What's the current status of this request? #262 has been closed, and @linknum23 mentioned above that once it has been sorted out, this would become much more flexible.

linknum23, 2024-06-17:
Given our progress on background streams / audio pipes Mopidy is much more feasible to integrate now. There is a lot integration work still needs to be done to add Mopidy to our system. At the moment we are focused on making our current feature set much more usable and robust. 

As for local audio hosted on the AmpliPi, the best supported option is running Logitech Media Server (LMS) on an AmpliPi and connecting a USB drive with audio files on it. An AmpliPi can be configured to run its own LMS and 4 LMS clients by enabling it in settings->Configuration->Logitech Media Server (LMS) Mode.

rtertiaer, 2024-06-25:
I'm extremely interested in this feature, as well as #355 . I'm going to briefly discuss inclusion of this in our 0.5 release, though I think it may be more realistic for 0.6, We have our scoping meeting at the end of this week :sunglasses: