micro-nova/AmpliPi #73: Add Server Sent Events (SSE)

linknum23, 2021-02-22:
SSE will allow notification on change events to be sent when the system state is modified. The idea is that a client registers to receive types of change events and is alerted when they happen. This will be useful for enhancing our webapp and should be  useful for many different types of home automation integration.

jberry, 2021-03-01:
We have a discussion going on (https://github.com/micro-nova/AmpliPi/discussions/77) talking about implementing some keypads.  How solid is the SSE decision that that's the route this will probably be going? 
Obviously anything can change/this isn't a given feature until it's delivered, but is SSE a solid-enough decision on the publish/subscribe system to use that it wouldn't be a waste of time for me to start learning/messing with it? (I've always been an MQTT user)

linknum23, 2021-03-01:
@jberry We have not invested any time in testing it yet, my only concern with it is whether or not the volume controls will feel really responsive with SSE, if they don't we might have to use websockets. This is something we should know in the next couple of weeks when we do a second pass on the webapp.

linknum23, 2021-03-30:
We are prototyping SSE by using it for updates, it's been pretty easy to work with so far.

linknum23, 2021-04-26:
Something to think about, we will probably need to add some sort of granular change support on the python side of things to avoid sending the full JSON state every time something is updated.

linknum23, 2023-12-21:
Some initial work was done in #254 on branch https://github.com/micro-nova/AmpliPi/tree/add_sse