micro-nova/AmpliPi #87: Add Tidal support?

mwbrink, 2021-03-10:
Would be great if Tidal could be added with HiTide.

https://github.com/shawaj/HiTide

linknum23, 2021-03-11:
It looks like the most up to date Linux Tidal support is from https://github.com/seniorgod/ifi-tidal-release (HiTide is based on this). Looking into this a little closer at ifi-tidal, Tidal support looks fairly easy to implement with that implementation. Here's what their command line interface looks like:
```
tidal_connect_application \
			--tc-certificate-path "/usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat" \
			--netif-for-deviceid eth0 \
			-f "DietPi stream to project“ \
			--codec-mpegh true \
			--codec-mqa false \
			--model-name "DietPi Streamer" \
			--disable-app-security false \
			--disable-web-security false \
			--enable-mqa-passthrough false \
			--playback-device "Project RS USB Audio 2.0: - (hw:1,0)" \
			--log-level 3
```
It should be rather easy to launch this, changing the playback device to ch0-ch3 and the model name to the specific stream's configuration. We have not tested this yet because it requires a Tidal account. @mwbrink I assume you have a Tidal account. Would you be willing to help us test this?

One thing to note is the underlying **tidal_connect_application** is only in closed source binary form, so this will need to be optionally downloaded and installed like a plugin.

linknum23, 2021-03-11:
Here are a list of steps that are needed to integrate TIDAL:
- [ ] Test tidal_connect_application on Linux
    - [ ] Can multiple tidal_connect clients run at once? Any special config besides name change? Do they need to have different accounts?
    - [ ] Is song/title/cover_art metadata easily available?
    - [ ] Can play/pause be controlled locally?
- [ ] Add tidal streaming class
- [ ] Add tidal connect install procedure (Make install optional?)

mwbrink, 2021-03-12:
I will do what I can to help.

eyeduck, 2021-03-18:
I would be interested in Tidal support also

linknum23, 2021-03-18:
Cool are either of you guys able to try out the `tidal_connect_application` on Linux? You can use the output of `aplay -L` to give you the audio output device strings available on your system. You can use this for the --playback-device parameter.

linknum23, 2021-04-22:
Is there anyone with an Tidal subscription to help us move forward on this?

mwbrink, 2021-12-20:
Okay, I've received my amplipi. I would like to have a go ath this, but need a little push to figure this out. I can ssh into the system. how do I best install the binaries?

linknum23, 2021-12-20:
The easiest way to start playing with it would be to use the HiTide installer that installs ifi-tidal-release and uses it in the HiTide service.

    curl -sSL https://raw.githubusercontent.com/shawaj/HiTide/main/install.sh | sudo bash

Then you should probably disable the service so you can play with with command line:

    sudo systemctl stop ifi-streamer-tidal-connect.service 
    sudo systemctl disable ifi-streamer-tidal-connect.service 

Looking at the service, its being run with the following comand line args:

    /usr/ifi/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --netif-for-deviceid eth0 -f AmpliPi-TiDAL --codec-mpegh true --codec-mqa false --model-name Pi-Streamer --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --playback-device ch1 --log-level 3

For now it looks like sources 2, 3, 4  (named 'ch1', 'ch2', 'ch3' respectively) can be used to playback audio. Source 1 is not being detected initially. Also it looks like the connect application is failing to create a socket on some unknown already used port.

To test on source 2 go to the the '2' tab in the app and select the blank input. This will configure the AmpliPi to use its Digital 2 input. Any zones connected to source 2 will play whatever TiDal is playing.

mwbrink, 2021-12-20:
it worked!
Got the same errors and some warnings on stream start as follows:
[2021-12-20 18:38:51.514] [tisoc] [error] [SpkconServer.cpp:173] socket bind error. fd=4 errorno=98 error=Address already in use
[2021-12-20 18:38:51.514] [tisoc] [error] [SpkconServer.cpp:54] already connected state:1
[2021-12-20 18:38:51.514] [tisoc] [critical] [SpeakerCommunicatorImpl.cpp:77] [ASSERT] start(ret)
STARTED TidalConnect services.
[2021-12-20 18:40:49.575] [tisoc] [warning] [PlaybackControllerImpl.cpp:472] Illegal state:1
[2021-12-20 18:40:50.252] [tisoc] [warning] [logger.cpp:22] [audio_worker.cpp:321] Attempting to initialize a decoder without initial data!

mwbrink, 2021-12-20:
On song change it spit out this error: 
[2021-12-20 18:43:19.372] [tisoc] [error] [logger.cpp:25] [operation_curl.cpp:258] curl_easy_perform failed:  Timeout was reached  -  Resolving timed out after 5513 milliseconds
[2021-12-20 18:43:19.372] [tisoc] [error] [logger.cpp:25] [http_stream_fetcher.cpp:162] Content-Range is missing from response!

mwbrink, 2021-12-20:
it's still playing

linknum23, 2021-12-21:
Can you try launching TiDAL on 2 different sources? Maybe ch1 and ch2? I'm assuming TiDAL allows multiple concurrent streams...

mwbrink, 2021-12-22:
I don't have a family account, thus I'm limited to one stream.

I restarted the unit to add an extender.  do I simply start the service or am I going to have to stop and disable the service?

mwbrink, 2021-12-22:
I tried running a second instance, but it didn't work. My  linux is poor, here is what I did:

pi@amplipi:~ $ /usr/ifi/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --netif-for-deviceid eth0 -f AmpliPi-TiDAL-ch2 --codec-mpegh true --codec-mqa fal
se --model-name Pi-Streamer --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --p
layback-device ch2 --log-level 3
STARTING TidalConnect services, version: 1.1.3
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
[... cut ...]
jack server is not running or cannot be started
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[2021-12-22 12:39:35.723] [tisoc] [error] [avahiImpl.cpp:184]  avahi_entry_group_add_service() err_code : -8
[2021-12-22 12:39:35.723] [tisoc] [error] [avahiImpl.cpp:144] MdnsServerImpl::createService() FAILED!
tidal_connect_application: thread-watch.c:162: avahi_threaded_poll_quit: Assertion `p' failed.
Aborted

linknum23, 2021-12-22:
Hmm looks like a duplicate name conflict:

Can you try the following in 2 separate ssh sessions. The commands use different names and outputs:

session1: 

    /usr/ifi/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --netif-for-deviceid eth0 -f AmpliPi-TiDAL1 --codec-mpegh true --codec-mqa false --model-name Pi-Streamer --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --playback-device ch2 --log-level 3

session 2:

    /usr/ifi/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --netif-for-deviceid eth0 -f AmpliPi-TiDAL2 --codec-mpegh true --codec-mqa false --model-name Pi-Streamer --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --playback-device ch3 --log-level 3

Verify that you can play to each of these, switching between them. No need to try simultaneous playback yet.

mwbrink, 2021-12-23:
The second command ran, always errors and aborts.

in this case session1 command was run second and output this:

pi@amplipi:~ $ /usr/ifi/ifi-tidal-release/bin/tidal_connect_application --tc-certificate-path /usr/ifi/ifi-tidal-release/id_certificate/IfiAudio_ZenStream.dat --netif-for-deviceid eth0 -f AmpliPi-TiDAL1 --codec-mpegh true --codec-mqa false --model-name Pi-Streamer --disable-app-security false --disable-web-security false --enable-mqa-passthrough false --playback-device ch2 --log-level 3
STARTING TidalConnect services, version: 1.1.3
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.front.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround41
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround50
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround51
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.surround71.0:CARD=0'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM surround71
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM iec958
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835_headpho.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3=2'
ALSA lib conf.c:4568:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5047:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM spdif
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
[... cut ...]
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
JackShmReadWritePtr::~JackShmReadWritePtr - Init not done for -1, skipping unlock
[2021-12-23 01:32:42.650] [tisoc] [error] [avahiImpl.cpp:184]  avahi_entry_group_add_service() err_code : -8
[2021-12-23 01:32:42.651] [tisoc] [error] [avahiImpl.cpp:144] MdnsServerImpl::createService() FAILED!
tidal_connect_application: thread-watch.c:162: avahi_threaded_poll_quit: Assertion `p' failed.
Aborted
pi@amplipi:~ $

Both commands seem to work on their own, but not together.

zjs, 2022-07-17:
I have a Tidal subscription and would be happy to help validate — is there something that'd be helpful for me to try? I can't quite tell if there was conclusion from the experiment above.

(There's also a [30 day trial](https://tidal.com/pricing) if being able to try things yourself would be easier.)

mwbrink, 2022-07-18:
The problem I had was my Tidal sub only allows 1 stream at a time, forbidding me to test concurrent streams.

charliesjc, 2022-07-21:
Unix socket /tmp/tisoc-controller has everything we need. It supplies all the now playing information and seems to control Tidal Connect. I can't use it without breaking the Tidal Service requiring a restart, though. I'm no expert in using sockets, maybe someone else knows what's going on?

linknum23, 2022-07-25:
That metadata looks very promising

charliesjc, 2022-07-25:
Indeed it is. I forgot to mention that the speaker_controller_application that's included with these binaries also uses the same socket because that also completely breaks when you connect to it.

SteveMicroNova, 2024-04-25:
I'd like to bump this thread and let everyone know that we're actively working on this now, hopefully we'll see this in an upcoming release in a reasonably short timeframe.

In the meantime, have any of you had luck continuing to use your custom implementations personally? I'm having troubles with that ifi-tidal-release repo, seems a few of the apt dependencies are deprecated and I may have to bark up a different tree.

darkm20, 2024-12-17:
ifi tidal, is a hacked version downloaded from ifi streamers, you should do the same implementation by contacting Tidal and ask for a developer account to get your place inside their authorized dealers / developers area.

SteveMicroNova, 2024-12-17:
> ifi tidal, is a hacked version downloaded from ifi streamers, you should do the same implementation by contacting Tidal and ask for a developer account to get your place inside their authorized dealers / developers area.

We're unlikely to get any special access like that due to being an open source project, but I can say we no longer intend to use ifi-tidal. We're now moving on to https://pypi.org/project/tidalapi/, which will require us to complete #415 as our current python version does not support that library

darkm20, 2024-12-17:
> We're unlikely to get any special access like that due to being an open source project, but I can say we no longer intend to use ifi-tidal. We're now moving on to https://pypi.org/project/tidalapi/, [...]

that's really a wise choice standing the situation! Good luck guys!