micro-nova/AmpliPi #92: Feature: Plexamp for Raspberry Pi

micronova-jb, 2021-03-16:
I've put some work into adding Plexamp support for the AmpliPi system - this issue will detail my findings.

- The current state of support:
As of right now, the AmpliPi system shows up as a local device to cast to from Plexamp/Plex on cellphone or desktop. This thread marks the beginning of my search for answers: https://forums.plex.tv/t/selecting-the-alsa-device-that-plexamp-outputs-to-on-raspberry-pi-cm3/692189

Thanks to odinb1's efforts (detailed here: https://github.com/odinb/bash-plexamp-installer) I was able to reach the current state. You can install plexamp on the Pi using their script, or you can follow the official install guide here: https://forums.plex.tv/t/plexamp-for-raspberry-pi-release-notes/368282

Be sure to include the workaround that OttoKerner linked in the post above. In either guide, you will have to edit the server settings @ /home/pi/.config/Plexamp/server.json. Odinb1 talks about this on their GitHub README.md - you'll have to get a working copy of server.json from an old desktop version of plexamp.

- What's missing?
Check out both of my forum posts ([1](https://forums.plex.tv/t/selecting-the-alsa-device-that-plexamp-outputs-to-on-raspberry-pi-cm3/692189) and [2](https://forums.plex.tv/t/configuration-settings-for-playing-audio-on-a-specific-channel-of-a-usb-audio-device-connected-to-raspberry-pi/696974)) to get the bigger picture of where we're at. Currently, there doesn't seem to be a way to specify channels for plexamp. This is problematic, given the fact that every other source we provide has the ability to specify a channel (which is set in asound.conf). The de-obfuscated javascript for the RPi version of plexamp can be found here: https://gist.github.com/micronova-jb/c7bc03ae046ce6ce988d3062555ed6e3

Perhaps there's something hidden in there that we haven't been able to find - for now, plexamp plays on ch1, ch2, and ch3 when cast to. These are the channels associated with the USB sound card.

I will try to keep this issue up to date as more information comes out!

linknum23, 2021-04-17:
@micronova-jb you should update this with the currentstate

micronova-jb, 2021-05-03:
https://files.plexapp.com/elan/plexamp-v1.0.5-pi.tar.gz

Adding the plexamp version we're running on the Pi

micronova-jb, 2021-05-03:
Seems to want to automatically run a plexamp instance with "AmpliPi-Template"
This should be removed

micronova-jb, 2021-05-03:
Notes on setup (port to documentation later):
Before running the deploy script, be sure to replace "server.json" in the streams directory with the file of the same name from the alternate plexamp install (odinb's RPi plexamp guide). This mainly ensures that [user][id] and [user][token] are properly captured, so you can simply take these values if you prefer. Without this step, plexamp won't work!

micronova-jb, 2021-05-03:
Need to make sure the streams.py line "plexamp_args = '/usr/bin/node /home/pi/plexamp/server/server.prod.js'" will actually work. When extracting the plexamp tar.gz, make sure the contents end up in /home/pi/plexamp/ (or we can change this line to a better location)

The current example actually allows you to pass a token and id if the proper one wasn't copied over. This is a bit more cumbersome, so proper copying is preferred (this just offers a server-side option)

closed 2021-06-24 by micronova-jb