micro-nova/AmpliPi #107: Feeature: Auto stop streaming source after ~1 min of all connected zones/groups being muted

linknum23, 2021-04-22:


bs42, 2021-05-07:
If implemented, please make this configurable, enable/disable the feature entirely and the timeout time for auto-stop.  I do not want a 1 minute auto stop, maybe a 10+ minute, but for my usage a 1 minute auto-stop could be annoying.

linknum23, 2021-05-07:
That is completely reasonable. This is mainly to handle the condition of "why did we leave Pandora playing all night at Jason's house".

I agree that a sane default with this would be 10min with the option to disable entirely. One other thing to note is this only makes sense for streams that we have direct control over like Pandora and Internet Radio. Many of the other streaming sources like Plex, Upnp,  Shairport, and Spotify Connect are controlled externally from AmpliPi.

linknum23, 2022-05-10:
We should also do something similar when all of the zones have been moved from a source.