micro-nova/AmpliPi #120: Detect audio level on analog inputs

Lohrer, 2021-05-18:
New preamp boards have "peak detection" circuits on them.

- [x] Read the output of the peak detect circuit on the Pi
- [ ] Fix the issue where when nothing is connected to the input, ~3.75V is present
- [ ] Auto mute, unmute?
- [ ] Setup actions based on detecting audio input on an analog source

Lohrer, 2021-05-18:
When no input is connected:
AL1_R rises to at least ~3.7V. It's hard to measure because it collapses when probed, but this is the value on C62.
The other side of C1 is at ~2.4V, so I don't know where the extra 1.3V+ comes from.

With an audio input connected:
The output of U25A matches the input exactly, except is clipped at 0V.
The cathode of D1 tends to spike high, then decay until something causes it to spike up again.

Lohrer, 2022-11-21:
The above comments are for old hardware and have been fixed. See [the preamp board's schematic](https://github.com/micro-nova/AmpliPi/blob/main/hw/sch/preamp_board/preamp_board_3a.pdf) page 2 for the circuit.

This is being worked on in the `peak-detect` branch. There is currently a skeleton `audiodetector.c` which does the actual ADC reading and outputs the status to a file. Amplipi's `ctrl.py` has been updated to read the playing/stopped status from that file.

To build the `audiodetector` binary, run `make -C amplipi/audiodetector` from the root of the AmpliPi repo. `./scripts/configure.py --audiodetector` will also build the binary and install the service.

Lohrer, 2022-12-21:
@jonahshader did you try the audiodetector on Preamp board rev 2.A?

Also, I tried from a fresh install and the rca_status file is generated but is empty. If it helps debug, I don't have anything playing and haven't had anything playing. Since the file is 0 bytes, I get the following error in AmpliPi:
> Error getting RCA audio status:
>    index out of range

linknum23, 2023-03-15:
Integrated this in 0.2.0

closed 2023-03-15 by linknum23