micro-nova/AmpliPi #141: Expansion Board Preamp Programming

micronova-jb, 2021-07-01:
We need to implement preamp-to-preamp flashing over UART.
UART pass through may work, if not a special firmware version may need to be implemented that knows how to program subsequent preamp boards. 

- [ ] Flash from master to first expansion preamp
- [ ] Flash multiple expansion units in a chain

Lohrer, 2021-08-16:
Expansion units will be programmed by having the previous Preamp Board enter "UART Passthrough" mode. The STM32's UART1 and UART2 will forward any messages received to the other. This adds a small latency to any reads, but doesn't seem to affect the `stm32flash` program. I have verified that this works for at least 1 master + 2 expansion units in the expansion_prog branch. There is still some software cleanup to do but the hardware has been verified functional.

Lohrer, 2021-09-13:
Expansion programming support added in PR #175

closed 2021-09-13 by Lohrer