micro-nova/AmpliPi #157: Verify firmware works with missing boards

Lohrer, 2021-08-11:
Firmware currently failts to initialize when no LED board is present. See if a missing power board causes the same issue, and then fix the firmware to note the error but continue initialization. Provide a status register or other way for the Pi to check the presence/absence of connected boards.

rtertiaer, 2024-04-19:
is this still valid, @Lohrer ?

Lohrer, 2024-04-19:
I have improved things since this issue was created (mostly by necessity for the various iterations of the power board hardware), but I know there's still at least 1 board that will cause issues if it's missing. I'm not sure which off the top of my head but next week I can do a quick check of which boards can be missing and which must be present. This really should be fixed, for the stability of the firmware but also to make things easier for production.