micro-nova/AmpliPi #158: Built-in Tests

linknum23, 2021-08-12:
Add built-in tests that can be run using the display. Initial tests: 

LED Board
- [ ] All on/scroll

Preout or Amplifier Board
- [ ] 6 Zone 50% volume/mute toggle

Preamp
- [ ] Program preamp and connected expansion board
- [ ] 4x2 sources on zone 1 (also zone 7 on expansion)
- [ ] 4 sources on all zones, toggle volume between 45 and 60%
- [ ] Poll/display 24V ADC + thermistors

Controller
- [ ] Aux/SPDIF
- [ ] Enet (ping micro-nova.com)

Manual (but instructions on screen)
- [ ] Desktop
- [ ] USB Power load
- [ ] Bootload

micronova-jb, 2021-09-08:
Will this be closed by #167 ?

linknum23, 2021-09-24:
Closed now that built-int-tests have been merged

closed 2021-09-24 by linknum23