micro-nova/AmpliPi #159: Update diagram and specifications

linknum23, 2021-08-12:
Several changes have been made to the system design since the start of the Kickstarter campaign. 

The major changes are:
- Updated design of the front and back panel,
- Added  preamplified outputs
- Added SPDIF and microphone inputs,
- Updated to a touch-enabled color TFT display.

What needs to be updated?
- [ ] https://github.com/micro-nova/AmpliPi/blob/master/docs/imgs/Signal_Diagram_with_Expansion_Out.png, https://github.com/micro-nova/AmpliPi/blob/master/hw/diagrams/signal-diagram.png
- [ ] https://github.com/micro-nova/AmpliPi/blob/master/hw/diagrams/amplipi-board-diagram.png
- [ ] https://github.com/micro-nova/AmpliPi/blob/master/hw/diagrams/high-level-diagram.jpg
- [ ] https://github.com/micro-nova/AmpliPi/blob/master/hw/diagrams/internals-diagram.jpg
- [ ] https://github.com/micro-nova/AmpliPi/blob/master/hw/diagrams/zone-expander-board-diagram.png
- [ ] Update website with above images and improve specification docs

Lohrer, 2022-06-01:
Mahdee updated the store page with the new pictures, anything else to be done here @linknum23?

closed 2022-06-01 by linknum23

brianhealey, 2022-06-01:
How can one tell which hardware version they have?

gorski123, 2022-06-01:
> How can one tell which hardware version they have?

@brianhealey Check here: https://github.com/micro-nova/AmpliPi/blob/main/hw/sch/hw_releases.md

In general, any main unit serial number > 128 will be the latest hardware rev.