micro-nova/AmpliPi #186: Audio Level Equalization

Lohrer, 2021-10-06:
Provide a way to equalize volumes across zones, DACs, and digital streams.
- Different rooms and/or speakers may have different perceived volumes so each zone needs to be adjustable independently.
- DAC 1 and DACs 2-4 output different voltage levels, so equalization should help normalize those.
- Different streams (Spotify, Pandora, etc) may have different output levels, so an option to adjust them should also exist.
- It might be a good idea to provide a way to equalize analog input volumes too, which would have to be applied at the volume controllers.

linknum23, 2022-08-22:
Note that we should also document that EQ can not be configured on a per zone basis with the current hardware.

iloveicedgreentea, 2023-07-21:
This is really important to me. Is there a way to do this even via manually editing files?

linknum23, 2023-07-24:
Assuming that you are talking about: 
>  Different streams (Spotify, Pandora, etc) may have different output levels, so an option to adjust them should also exist.

We understand this is a very useful feature, sorry it doesn't exist yet!

That said, there are some ways to do this manually, per stream, but most of the streams have such simple configuration files that their generation was built into [amplipi/streams.py](blob/main/amplipi/streams.py). So at the moment this will require modifying the config generation code corresponding to each stream.

iloveicedgreentea, 2023-07-28:
Ah actually I was talking about EQ not loudness equalization. I misread the op. Is eq possible at all?

linknum23, 2023-08-08:
EQ is possible but only per source, check out another users exploration of it here: #308 

Adding per zone EQ is something we will consider with newer hardware revisions, it is not possible without changes to the hardware.