micro-nova/AmpliPi #199: Allow HDMI as input

stoppablemurph, 2021-10-29:
I'm not sure if it's even physically possible, and definitely not sure how difficult it would be to do if it is, but it would be nice to be able to use the HDMI out port as an input for something like a Chromecast (since I'm only just now realizing Chomecast Audio was discontinued and there's not really anything else quite like it that I've been able to find).

Obviously native cast support would probably be even better for this specific use case, but I'm sure there are plenty of other use cases where someone might want to use the HDMI port as an input.

linknum23, 2021-10-30:
Our current hardware only supports using the HDMI port as an output. Have you thought of using an HDMI to RCA converter to convert the output of the Chromecast to one of AmpliPi's RCA inputs?

stoppablemurph, 2021-10-30:
No worries if it's not possible. I think an HDMI audio extractor is probably my best bet.

closed 2021-10-30 by stoppablemurph