micro-nova/AmpliPi #200: plexamp not working out of the box

stoppablemurph, 2021-10-30:
i have been unable to get plexamp working correctly as an input option. plex apps themselves correctly identify the plexamp created by the amplipi as an output option, but when i try to play anything, nothing happens. the player just sits at 00:00 for several seconds before giving up like i never hit play, with no errors or anything. (playing directly from the device speakers works fine, so it doesn't seem to be an issue with my plex server setup)

i tried digging around for logs a bit. closest i found to anything helpful was

```
$ cat /home/pi/amplipi-dev/config/srcs/1/.config/Plexamp/mpd.log
Oct 29 19:38:34.324: exception: OutputThread could not get realtime scheduling, continuing anyway: sched_setscheduler failed: Operation not permitted
Oct 29 19:38:34.325: client: [0] opened from 127.0.0.1:57876
Oct 29 19:42:47.814: exception: OutputThread could not get realtime scheduling, continuing anyway: sched_setscheduler failed: Operation not permitted
Oct 29 19:42:47.820: client: [0] opened from 127.0.0.1:39886
Oct 30 12:47:38.712: exception: OutputThread could not get realtime scheduling, continuing anyway: sched_setscheduler failed: Operation not permitted
Oct 30 12:47:38.718: client: [0] opened from 127.0.0.1:40792
```

here's the plexamp server.json generated by the amplipi ui. i'm not sure what the "server" and "state" fields are supposed to look like, but being "null" is a little suspicious.

```
$ cat /home/pi/amplipi-dev/config/srcs/1/.config/Plexamp/server.json
{
  "player": {
    "name": "ampliplex",
    "identifier": "REDACTED"
  },
  "user": {
    "token": "REDACTED"
  },
  "state": "null",
  "server": "null",
  "audio": {
    "normalize": "false",
    "crossfade": "false",
    "mpd_path": "/home/pi/amplipi-dev/config/srcs/1/mpd.conf"
  }
```

stoppablemurph, 2021-10-30:
it may be worth noting that i don't have a plex pass subscription. is that a prerequisite to this feature functioning? if so, it might be a good idea to make that more clear and prevent the user from creating the input or something.

linknum23, 2021-10-31:
If I recall correctly PlexAmp is a paid feature. I know we had to upgrade our account before we could start testing support for it. Let me verify this is actually the case, but I'm guessing we will need to add this as a note to the docs.

stoppablemurph, 2021-10-31:
After purchasing a Plex Pass subscription and creating a new plexamp input source in the amplipi, it's still behaving the same as it was before.

Lohrer, 2021-11-01:
I tried connecting with my (plexpass) account and am getting what I assume is the same issue. I setup a new Plexamp stream, then set it as the stream for source 1. When using "select player" on Plex Web, all appears OK but as you describe once I tried to play anything it gets stuck at 00:00 for a few seconds before giving up. Using the Plexamp Android app I get the error "Could not switch to player".

Sorry you've encountered an issue here, there's definitely something going wrong and we will have to investigate further.

Lohrer, 2022-03-21:
Looks like Plexamp 4 brings back remote control, so we may be able to get a headless version working again.
https://forums.plex.tv/t/plexamp-release-notes/221280/38

linknum23, 2022-03-21:
Yeah it looks like it is getting pretty close: although time will tell if it is only released as a 64-bit binary forcing us to update to a 64-bit OS.

Here's where I found the latest news on the pi integration:

- https://forums.plex.tv/t/plexamp-headless-on-a-pi-oled-screen-compatibility/778060/8 (this is the comment Elon mentions 64-bit)
- https://forums.plex.tv/t/future-of-plexamp-on-raspberry-pi/717577/106

linknum23, 2022-05-04:
Looks like there is a prototype of a headless plexamp release out there, main issue is that it is on a 64-bit pi (AmpliPi currently expects a 32-bit os).
Here's a link to the release and the thread: 
- Release: https://plexamp.plex.tv/headless/Plexamp-Linux-arm64-v4.2.2-beta.1c.tar.bz2
- Thread: https://forums.plex.tv/t/future-of-plexamp-on-raspberry-pi/717577/161

rickballard, 2022-05-07:
I am concerned that future AmpliPi's input sources may require an increasing level of technical know-how.  Will non-technical AmpliPi-ers who can barely manage their own Plex Servers, be able to set-up a headless Plexamp input?  Perhaps headless Plexamp could be packaged to work as a wizard that self-installs/configures through an "optional add-ons menu", in the AmpliPi web-app?

Lohrer, 2022-06-02:
Blocked by #335