micro-nova/AmpliPi #207: FEATURE REQUEST : HDRadio / NRSC-5 support

dlasher, 2021-11-01:
Looking forward AmpliPi shipping - wanted to ask to include HDRadio support. ( https://github.com/theori-io/nrsc5)

Although I would recommend the often-updated fork: https://github.com/markjfine/nrsc5-dui

Thanks!

linknum23, 2021-11-02:
This looks like it could be an upgrade from our current FM radio support that already uses the same dongle. I will add some notes below for when we get time to work on this (or in the awesome case anyone would like to help).

The underlying decoder appears to be **nrsc5**, we would have to build this for the pi. As for using the decoder the gui that you referenced **nrsc5-dui** has a lot of good example code that would speed up the integration time. We would basically just need to implement this function and everything it calls to keep track of state and song information: https://github.com/markjfine/nrsc5-dui/blob/cd0b378276fb9d686d2f2fe8f24a82f0eba25b4e/nrsc5-dui.py#L918

dlasher, 2021-11-02:
Sweet! Thanks for taking a look, appreciate it!