micro-nova/AmpliPi #211: Feature: Add Zone and Group Based Views

linknum23, 2021-11-08:
A Zone or Group based view would naturally be easier to see and control what is playing on a specific room/zone and would make the perfect interface for a room's touch panel controls. The UI would require minimal changes, since an initial prototype would look very similar to an input with only one zone connected.

Some things need to be figured out:
- What happens when a new input is selected when other zones are also connected to that input?
  - Option 1: Split from the other zones which will stay connected to the old input. The new input only plays on this zone.
  - Option 2: Keep grouping and change all zones to the new input 
- What happens when a new input is selected and it is already being played on other zones?
  - Option 1: Steal the input from the other zones. The new input only plays on this zone.
  - Option 2: Add this zone to the grouping of zones that is playing the selected input. 
- Do we show the other grouped zones? How?

From a low level API standpoint there would only need to be one change we would have to add the concept of a Zone not being connected to a source.  Basically if its not connected the zone would need be muted but we would need to keep track of this state.

linknum23, 2021-11-09:
@sumnerboy Is this something you would be interested in testing and giving us feedback on? I could probably get a prototype of this working that you could play with sometime in the next could of weeks.

sumnerboy12, 2021-11-09:
Absolutely!

linknum23, 2021-11-12:
After talking through this with @Lohrer I made some initial concepts, using a rotated tablet view to emulate a wall touch panel. We came up with the concept of embedding all of the different use cases in the selection bar. 

Here's a mock up of the simplified interface:
[image]

Here's a potential drop down with options to join other currently playing audio sources or connect to a new one:
[image]

(This assumes that the zone is the only thing connected to the current input, if it wasnt we would need the option to split away from other zones connected to the same input)

I hope this makes sense. Its very early on.

sumnerboy12, 2021-11-14:
Looks like a good start. 

Do we need to distinguish between a currently playing source and a new one? When thinking about it from a zone POV - do we really care if another zone is already listening to the input we select?

Also , personally I am not a fan of drop down lists with empty and ***header*** items - I think it is confusing and almost looks like there is a bug in the list rendering. Everything in a drop down list should be selectable IMO. 

If we have to have the distinction then what about something like;
```
* TV - rca [playing]
* Aleshas - spotify [playing]
* AmpliPi - shairport
* Matt & Kim - pandora
* AmpliPi - dlna
```

Just a couple of thoughts off the top of my head :).

kbromer, 2022-01-13:
Wanted to add a vote for this feature and also volunteer to test if that's helpful.  

Love the early work here - one bit of humble feedback that immediately pops to mind is wanting to maintain UI consistency.  With the existing webapp, you've established the "top thing is static, you select the bottom thing" paradigm.  (i.e. the input is static, you're shuffling the zones underneath it).  In this inverted setup, we'll want to hold the top thing static (the zone) and put the selector underneath.  

Its a subtle/minor thing, but may make it more intuitive to the existing layout.

linknum23, 2022-01-14:
Thanks for the feedback! We really need to circle back on this soon.

bs42, 2022-01-18:
As I get used to using the interface as it is, one thing I have come to understand is why ya'll chose an input centric model for the interface.  With the unit being "limited" to the 4 input streams, digital or analog, I can see why basing the system on inputs was done.  

If the interface changes to be Zone centric it could confuse the issue that one cannot play more than 4 streams across 6+ zones.

I don't have a solution, just wanted to offer that point of view.

linknum23, 2022-01-18:
The initial intent with this is to handle the use case of having a
simplified ui for a room or group of rooms controlled by a wall mounted
tablet. I agree that it will obscure the 4 input restriction, but we would
like to try it out and see what makes sense for this.