micro-nova/AmpliPi #218: Web UI returning 500 HTTP code after a few days sitting idle

sumnerboy12, 2021-11-29:
I have a few internet streams running on 3 of my inputs, and a Spotify account setup on the other input, but not playing anything. After a few days the web UI fails to load and returns a 500 error.

I have to restart my AmpliPi to get the UI back.

I am unable to SSH into the device either; 

```
ben@HIHI:~$ ssh pi@amplipi
ssh_exchange_identification: Connection closed by remote host
```

linknum23, 2021-11-29:
Hmm I haven't seen this issue on a normal install. So sorry that your having issues.

The initial guess I have is that there is something that is generating too many logs and filling up the storage. We have seen this in some weird test configurations.

For ssh you need to add the '.local' suffix to the address so the command `ssh pi@amplipi.local` should work, password is the pi default: 'raspberry'. When you are ssh'd in check the disk usage with the command `df -h` the use % will be around 30% or so.

sumnerboy12, 2021-11-29:
I have a local dns server running with an entry for my amplipi, so `ssh pi@amplipi` works on my LAN. I tried `amplipi.local` just in case and got the same result trying to SSH in. So whatever has happened has borked SSH as well.

Before I hard reboot is there anything else you would like me to try to diagnose what might be going wrong?

Lohrer, 2021-11-29:
Can you check the disk % on the display (tap to wake up)? Or is that not working either?

sumnerboy12, 2021-11-29:
[image]

linknum23, 2021-11-29:
Hmm you will need to reboot and look at the logs over ssh. After rebooting can you ssh in and send us the output of `journalctl --user-unit amplipi > amplipi_debug.log`.

sumnerboy12, 2021-11-29:
```-- Logs begin at Thu 2019-02-14 05:11:59 EST, end at Mon 2021-11-29 15:26:48 EST. --
Nov 26 06:45:07 amplipi systemd[534]: Started Amplipi Home Audio System.
Nov 26 06:45:19 amplipi authbind[805]: vollibrespot v0.2.4 61563b7 2021-07-22 (librespot 08d8bcc 2020-10-07) -- Built On 2021-07-22
Nov 26 06:45:19 amplipi authbind[805]: Reading Config from "config.toml"
Nov 26 06:45:19 amplipi authbind[805]: [Vollibrespot] : Using Alsa backend with device: ch0
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Ben disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Ben connected to 0
Nov 26 06:45:20 amplipi authbind[805]: National Radio disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting National Radio to 1...
Nov 26 06:45:20 amplipi authbind[805]: National Radio connected to 1
Nov 26 06:45:20 amplipi authbind[805]: Breeze disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting Breeze to 2...
Nov 26 06:45:20 amplipi authbind[805]: Breeze connected to 2
Nov 26 06:45:20 amplipi authbind[805]: Hauraki disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting Hauraki to 3...
Nov 26 06:45:20 amplipi authbind[805]: Hauraki connected to 3
Nov 26 06:45:20 amplipi authbind[805]: Preamp found at address 8
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Ben disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Renee disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Olivia disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Holly disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Toby disconnected
Nov 26 06:45:20 amplipi authbind[805]: National Radio disconnected
Nov 26 06:45:20 amplipi authbind[805]: Hauraki disconnected
Nov 26 06:45:20 amplipi authbind[805]: Breeze disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Ben disconnected
Nov 26 06:45:20 amplipi authbind[805]: AmpliPi - Ben connected to 0
Nov 26 06:45:20 amplipi authbind[805]: National Radio disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting National Radio to 1...
Nov 26 06:45:20 amplipi authbind[805]: running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://radionz-ice.streamguys.com/national.mp3', 'ch1', '--song-info', '/home/pi/amplipi-dev/config/srcs/1/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/1/log']
Nov 26 06:45:20 amplipi authbind[805]: National Radio (stream: http://radionz-ice.streamguys.com/national.mp3) connected to 1 via ch1
Nov 26 06:45:20 amplipi authbind[805]: Breeze disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting Breeze to 2...
Nov 26 06:45:20 amplipi authbind[805]: running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps', 'ch2', '--song-info', '/home/pi/amplipi-dev/config/srcs/2/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/2/log']
Nov 26 06:45:20 amplipi authbind[805]: Breeze (stream: https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps) connected to 2 via ch2
Nov 26 06:45:20 amplipi authbind[805]: Hauraki disconnected
Nov 26 06:45:20 amplipi authbind[805]: connecting Hauraki to 3...
Nov 26 06:45:20 amplipi authbind[805]: running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8', 'ch3', '--song-info', '/home/pi/amplipi-dev/config/srcs/3/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/3/log']
Nov 26 06:45:20 amplipi authbind[805]: Hauraki (stream: http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8) connected to 3 via ch3
Nov 26 06:45:20 amplipi authbind[805]: INFO:     Started server process [805]
Nov 26 06:45:20 amplipi authbind[805]: INFO:     Waiting for application startup.
Nov 26 06:45:20 amplipi authbind[805]: INFO:     Application startup complete.
Nov 26 06:45:20 amplipi authbind[805]: INFO:     Uvicorn running on http://0.0.0.0:80 (Press CTRL+C to quit)
Nov 26 06:45:20 amplipi authbind[805]: INFO:     192.168.10.21:34844 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:20 amplipi authbind[805]: INFO:     127.0.0.1:53984 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:21 amplipi authbind[805]: INFO:     127.0.0.1:53992 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:22 amplipi authbind[805]: [7332e5e0] mpeg4audio demux packetizer: AAC channels: 2 samplerate: 48000
Nov 26 06:45:22 amplipi authbind[805]: INFO:     127.0.0.1:53998 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:23 amplipi authbind[805]: [73302d30] prefetch stream error: unimplemented query (264) in control
Nov 26 06:45:23 amplipi authbind[805]: INFO:     127.0.0.1:54004 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:24 amplipi authbind[805]: [Vollibrespot] : couldn't parse packet from 192.168.10.45:5353: type 47 is invalid
Nov 26 06:45:24 amplipi authbind[805]: INFO:     127.0.0.1:54010 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:25 amplipi authbind[805]: INFO:     127.0.0.1:54016 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:26 amplipi authbind[805]: [7560cbb0] mpeg4audio demux packetizer: AAC channels: 2 samplerate: 22050
Nov 26 06:45:26 amplipi authbind[805]: INFO:     127.0.0.1:54022 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:27 amplipi authbind[805]: INFO:     127.0.0.1:54026 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:28 amplipi authbind[805]: INFO:     127.0.0.1:54030 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:29 amplipi authbind[805]: INFO:     127.0.0.1:54034 - "GET /api HTTP/1.1" 200 OK
[... cut ...]
Nov 26 06:45:34 amplipi authbind[805]: INFO:     127.0.0.1:54056 - "GET /api HTTP/1.1" 200 OK
Nov 26 06:45:35 amplipi authbind[805]: INFO:     127.0.0.1:54060 - "GET /api HTTP/1.1" 200 OK
Nov 29 15:26:43 amplipi authbind[805]: INFO:     127.0.0.1:54064 - "GET /api HTTP/1.1" 200 OK
Nov 29 15:26:47 amplipi authbind[805]: INFO:     192.168.10.21:35010 - "GET /api HTTP/1.1" 200 OK
Nov 29 15:26:48 amplipi authbind[805]: [0162b560] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
```

linknum23, 2021-11-29:
Hmm there is something missing in these log files. Most of the log is from a couple of days ago. We might be able to debug this further by running the debug webserver. It will output everything it is doing to the screen.

    ./amplipi-dev/scripts/run_debug_webserver

sumnerboy12, 2021-11-29:
Do I need to shutdown the current web server first?

linknum23, 2021-11-29:
no, edited above explanation

sumnerboy12, 2021-11-29:
```
pi@amplipi:~ $ ./amplipi-dev/scripts/run_debug_webserver
librespot: no process found
pianobar: no process found
shairport-sync: no process found
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
kill: usage: kill [-s sigspec | -n signum | -sigspec] pid | jobspec ... or kill -l [sigspec]
Preamp found at address 8
AmpliPi - Ben disconnected
AmpliPi - Ben disconnected
vollibrespot v0.2.4 61563b7 2021-07-22 (librespot 08d8bcc 2020-10-07) -- Built On 2021-07-22
Reading Config from "config.toml"
[Vollibrespot] : Using Alsa backend with device: ch0
AmpliPi - Ben connected to 0
National Radio disconnected
National Radio disconnected
connecting National Radio to 1...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://radionz-ice.streamguys.com/national.mp3', 'ch1', '--song-info', '/home/pi/amplipi-dev/config/srcs/1/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/1/log']
National Radio (stream: http://radionz-ice.streamguys.com/national.mp3) connected to 1 via ch1
Breeze disconnected
Breeze disconnected
connecting Breeze to 2...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps', 'ch2', '--song-info', '/home/pi/amplipi-dev/config/srcs/2/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/2/log']
Breeze (stream: https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps) connected to 2 via ch2
Hauraki disconnected
Hauraki disconnected
connecting Hauraki to 3...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8', 'ch3', '--song-info', '/home/pi/amplipi-dev/config/srcs/3/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/3/log']
Hauraki (stream: http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8) connected to 3 via ch3
Preamp found at address 8
AmpliPi - Ben disconnected
AmpliPi - Renee disconnected
AmpliPi - Olivia disconnected
AmpliPi - Holly disconnected
AmpliPi - Toby disconnected
National Radio disconnected
Hauraki disconnected
Breeze disconnected
AmpliPi - Ben disconnected
AmpliPi - Ben disconnected
vollibrespot v0.2.4 61563b7 2021-07-22 (librespot 08d8bcc 2020-10-07) -- Built On 2021-07-22
Reading Config from "config.toml"
[Vollibrespot] : Using Alsa backend with device: ch0
AmpliPi - Ben connected to 0
National Radio disconnected
National Radio disconnected
connecting National Radio to 1...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://radionz-ice.streamguys.com/national.mp3', 'ch1', '--song-info', '/home/pi/amplipi-dev/config/srcs/1/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/1/log']
National Radio (stream: http://radionz-ice.streamguys.com/national.mp3) connected to 1 via ch1
Breeze disconnected
Breeze disconnected
connecting Breeze to 2...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps', 'ch2', '--song-info', '/home/pi/amplipi-dev/config/srcs/2/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/2/log']
Breeze (stream: https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps) connected to 2 via ch2
Hauraki disconnected
Hauraki disconnected
connecting Hauraki to 3...
running: ['/home/pi/amplipi-dev/venv/bin/python', '/home/pi/amplipi-dev/streams/runvlc.py', 'http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8', 'ch3', '--song-info', '/home/pi/amplipi-dev/config/srcs/3/currentSong', '--log', '/home/pi/amplipi-dev/config/srcs/3/log']
Hauraki (stream: http://ais-nzme.streamguys1.com/nz_009/playlist.m3u8) connected to 3 via ch3
INFO:     Started server process [2148]
INFO:     Waiting for application startup.
INFO:     Application startup complete.
INFO:     Uvicorn running on http://0.0.0.0:5000 (Press CTRL+C to quit)
AmpliPi zeroconf - registering service: ServiceInfo(type='_http._tcp.local.', name='amplipi-api._http._tcp.local.', addresses=[b'\xc0\xa8\n-'], port=5000, weight=0, priority=0, server='amplipi.local.', properties={'path': '/api/', 'name': 'AmpliPi', 'vendor': 'MicroNova', 'version': '0.1.7-alpha.0', 'web_app': 'http://amplipi.local:5000', 'documentation': 'http://amplipi.local:5000/doc'}, interface_index=None)
INFO:     127.0.0.1:45488 - "GET /api HTTP/1.1" 200 OK
AmpliPi zeroconf - finished registering service
INFO:     127.0.0.1:45494 - "GET /api HTTP/1.1" 200 OK
[733212a8] mpeg4audio demux packetizer: AAC channels: 2 samplerate: 48000
INFO:     127.0.0.1:45502 - "GET /api HTTP/1.1" 200 OK
[73305f68] prefetch stream error: unimplemented query (264) in control
[... 51 lines cut ...]
INFO:     127.0.0.1:45724 - "GET /api HTTP/1.1" 200 OK
INFO:     127.0.0.1:45730 - "GET /api HTTP/1.1" 200 OK
INFO:     127.0.0.1:45734 - "GET /api HTTP/1.1" 200 OK
INFO:     127.0.0.1:45738 - "GET /api HTTP/1.1" 200 OK
INFO:     127.0.0.1:45742 - "GET /api HTTP/1.1" 200 OK
INFO:     127.0.0.1:45746 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53885 - "GET / HTTP/1.1" 200 OK
INFO:     192.168.10.25:53885 - "GET /static/css/style.css HTTP/1.1" 200 OK
INFO:     192.168.10.25:53885 - "GET /static/js/script.js HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /static/js/settings.js HTTP/1.1" 200 OK
INFO:     192.168.10.25:53887 - "GET /None HTTP/1.1" 422 Unprocessable Entity
[... cut ...]
INFO:     192.168.10.25:53884 - "GET /static/css/libs/google-fonts/opensans/v18/mem5YaGs126MiZpBA-UN_r8OUuhp.woff2 HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
INFO:     192.168.10.25:53884 - "GET /api HTTP/1.1" 200 OK
```

sumnerboy12, 2021-11-29:
The requests from 192.168.10.25 (at the end) was me opening a browser on my PC and pointing at http://amplipi:5000.

sumnerboy12, 2021-11-29:
The following just appeared...

```
[01f4fb40] main input error: ES_OUT_SET_(GROUP_)PCR  is called too late (pts_delay increased to 1000 ms)
[01f14290] alsa audio output error: cannot estimate delay: Broken pipe
```

linknum23, 2021-11-29:
Those are outputs from vlc (the backend for the internet radio). I don't think they are related to this issue.

linknum23, 2021-11-29:
Thinking about this from the happens every 3 days perspective it sounds like a very slow memory leak. Could you attach a copy of your config? app->settings->Configuration and Reset->Download

If you have any Pandora accounts in the config please remove the password before sending,

sumnerboy12, 2021-11-29:
Can I email this to you?

linknum23, 2021-11-29:
lincoln@micro-nova.com

linknum23, 2021-11-29:
Got it! I'll test running all of those streams overnight and see if there's anything we notice on our side.

linknum23, 2021-11-30:
Hmm no real changes overnight, only using +2MB which seems insignificant. We'll keep this running and check on it again tomorrow. If this ends up being a slow leak in vlc, one fix may be to implement something like #107 after 15-20 minutes.

sumnerboy12, 2021-11-30:
Yep same here - I shut down the debug web server and rebooted my device and it has been running for 24 hours - all fine currently...

sumnerboy12, 2021-12-03:
Been running fine here...

linknum23, 2021-12-03:
Same. We are going to keep an eye of the memory footprint over time on our main AmpliPi but I haven't seen any leaks from internetradio/vlc or spotify/vollibresport using your config over 4 days.
I think we need more data points on this to figure this out.
To debug this issue further we may need to be able to view the system in a broken state. If SSH doesn't work when it is in this failure state, you will need to connect a keyboard and monitor to the system and restart it.

sumnerboy12, 2021-12-08:
Been checking a couple of times every day, and just now I looked and it is down. Yours?

linknum23, 2021-12-08:
Oh no! We have been developing lots of new code on our AmpliPis so we had to stop running that test after 5 days.

Can you get into your system via ssh or attached hdmi display?

sumnerboy12, 2021-12-08:
Nope - once it goes down like this any connection over the network seems 
impossible.

Could I try a USB connection via the service port on the back?

linknum23, 2021-12-09:
The service port can only be made available by the pi's bootloader. So it will need a restart....

Sorry we haven't figured this out yet, we've been trying to make a new software release with new firmware and zone expander support. And we've been building a better software updater to handle all of that. We needed the AmpliPi's we were using to test your configuration for testing all of that.

We are building up a stock AmpliPi today to just put on our network and replicate your exact configuration. We are hoping this will give us more insight into what is going wrong. This will also allow us to verify that this issue is being caused by something in the AmpliPi software.

linknum23, 2021-12-13:
Sadly we lost power over the weekend, here. This morning we put your mirror AmpliPi on battery backup power, still trying to reproduce this, but it is going to take a couple more days...

linknum23, 2021-12-16:
Your AmpliPi has been up for 3 days 2 hours. Still going strong for now.

sumnerboy12, 2021-12-16:
Hmmm, mine has frozen up again, after upgrading to 0.17.0 (it came with
0.17.0-beta or something).

linknum23, 2021-12-16:
How long after the update did it freeze? Same symptoms as before?

sumnerboy12, 2021-12-16:
A couple of days, yep same symptoms.

linknum23, 2021-12-16:
Hmm, could you try re-imaging your AmplPi? Just in case there is something funky with your image.

 We posted a guide on discourse: https://amplipi.discourse.group/t/re-imaging-the-raspberry-pi/46. It has the latest software on it.

linknum23, 2021-12-16:
Thanks again for helping us troubleshoot this @sumnerboy12, we really want to get you a working solution here.

sumnerboy12, 2021-12-16:
Thank you for the great support! Will try re imaging and testing again.

linknum23, 2021-12-19:
There are preliminary reports that we were able to reproduce this (from Jason). Well check it out tomorrow.

linknum23, 2021-12-21:
We checked this out most of yesterday, looking deep into the logging that we added. This looks a lot like an underlying problem in the raspbian version we are using. My guess is that this is invoked somehow by the multiple long running network streams.

Here are the symptoms of an AmpliPi in this broken state:
- no SSH access, ssh attempts return: `ssh_exchange_identification: read: Connection reset by peer`
- API is still responsive, but all webpages return HTTP:500 errors
- display touch does not "wake" display view
- completely missing logfiles, nothing was logged since days before (logging stopped working before display and ssh)

sumnerboy12, 2021-12-21:
Sounds like a nasty one! Let me know if there is anything I can do to help with testing.

Lohrer, 2022-01-03:
Based on https://www.raspberrypi.com/documentation/computers/os.html#how-to-get-back-to-safety I updated just the kernel and related packages of our testing AmpliPi: `sudo apt install libraspberrypi0 libraspberrypi-{bin,dev,doc} raspberrypi-bootloader raspberrypi-kernel`. Newer Raspberry Pi OSs have added PulseAudio and updating the whole OS breaks our ALSA setup, so for now I think the kernel update is the best thing we can test with.

We will let this run and see if it breaks again...

linknum23, 2022-01-17:
Happy to announce that fixing the kernel has allowed us to debug this a lot further.

After 6 days AmpliPi is still up and running but 2 of the internet radio streams are in a bad state.

On initial inspection, it looks like vlc is getting into a bad state and when it gets into a bad state the internet radio plugin starts logging the failed state as fast as possible. We will need to investigate further to get a fix for this, but we are working toward resolving this.

Next strep is to add some code to detect this state and restart the stream as needed.

sumnerboy12, 2022-01-17:
Sounds promising - great work team!

sumnerboy12, 2022-02-11:
Any progress with resolving this? I am still getting a lock up every week or so, requiring a hard restart.

linknum23, 2022-02-14:
Not yet. We've been meaning to get back to this but haven't yet. Our next release will at the very least have the mitigation for this bug (which is don't keep logging vlc failures after they happen once).

linknum23, 2022-03-29:
I've been iterating on some simple changes to the script that underlies the internet radio stream, runvlc.py. Any chance you want to test it out?

1. Download the updated runvlc.py script:
https://raw.githubusercontent.com/micro-nova/AmpliPi/internet_radio_fail/streams/runvlc.py
1. SCP it over with ```scp runvlc.py pi@amplipi:amplipi-dev/streams/ ```
1. For each source (tab) playing an internet radio stream select the empty stream (``              ``), then re-select the radio station. This will restart script controlling the playback.

sumnerboy12, 2022-03-30:
Done, done and done. :)

linknum23, 2022-03-30:
Awesome! Please let me know how it treats ya. I was using the following command to check all of the logs every minute:

```watch -n 60 ssh pi@amplipi tail amplipi-dev/config/srcs/*/log````

sumnerboy12, 2022-04-04:
No change unfortunately. System has frozen after a few days.

linknum23, 2022-04-04:
Ours hasn't frozen, but all 3 radio stations stopped playing over the weekend. I just updated the script to completely reset more of the underlying vlc instance. I'll see if this fix does any better.

sumnerboy12, 2022-04-07:
Ok - updated to the latest script and restarted all my radio streams. This issue has been ongoing now for over 4 months. My AmpliPi is barely usable since most times I try to use it, I find it is locked up.

Can this please be bumped up the priority list?

linknum23, 2022-04-09:
I am so sorry, this bug is hard to test since it takes 3-7 days to appear. Since Jan we didn't have enough hardware to even test this bug on. Over the last 3 weeks we've been able to get enough hardware to dedicate one of the AmpliPi controllers to testing this again. 

We are still planning on getting this fix in the next release unless it requires a kernel upgrade to fully fix. Our updater doesn't have support for kernel upgrades yet. If that last fix doesn't fix it we will add the details on how to do a kernel upgrade so at the very least it can be fixed for you on your side.

I was very much hoping that we could work around the issue by adding better restart logic to our VLC script since the issue seems very correlated with internet radio stations, specifically https://tunein-icecast.mediaworks.nz/breeze_cantebury_128kbps seems to have the most issues and seems to get stuck in a bad state reconnecting.

linknum23, 2022-04-11:
## Updating the kernel:

1. SSH into the pi and run the following command
`sudo apt update && sudo apt install libraspberrypi0 libraspberrypi-{bin,dev,doc} raspberrypi-bootloader raspberrypi-kernel && sudo reboot
`

If all goes well it should update the kernel and restart to run the new kernel.

Our internet radio development box just got a kernel update and we are testing out your configuration on it now. We really hope this kernel update along with the restart on failure capabilities of the internet radio stream fixes your issues.

sumnerboy12, 2022-04-11:
Ok - kernel updated and rebooted. Will keep an eye on things and report back if there are any issues. Thanks!

linknum23, 2022-04-11:
Sweet. Thanks for all of the help testing!

linknum23, 2022-04-19:
Hows it working for you? On my box after several days breeze radio stopped working and got stuck in a bad state but after 8 days the rest of the system is still going strong. I restarted breeze today and everything is still working good.

From what I can tell the vlc restart logic added helps extend the playback of problematic internet radio streams but some of those stations will require a full restart when they get in a bad state. We will need to troubleshoot this more in the future.

sumnerboy12, 2022-04-19:
Yeah so far so good. No lockups in 8 days so it certainly seems improved - thanks so much!

linknum23, 2022-04-19:
Sweet. I just added that kernel upgrade to the update process. Hoping to pull all that into a release soon.

linknum23, 2022-04-25:
Closing this! With #317 and #297 this should be resolved.

closed 2022-04-25 by linknum23