micro-nova/AmpliPi #234: Upgrade to newer Raspberry Pi OS

Lohrer, 2021-12-16:
Currently we use `2020-08-20-raspios-buster-armhf` as our base Raspberry Pi OS image. With either a newer image or with an `apt upgrade` the AmpliPi audio with ALSA stops working. Either we need to switch away from directly using ALSA or we need to fix our setup to work with newer Raspberry Pi OS audio setups.

brianhealey, 2022-06-01:
FYI.  Using 0.1.8 I was able to run apt update without noticeable issues.

closed 2022-06-02 by Lohrer

brianhealey, 2022-06-02:
I did.  `update`, `upgrade`, and `dist-update`.  From what I can tell everything is working.

`pi@amplipi:~ $ sudo apt dist-upgrade
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Calculating upgrade... Done
The following packages were automatically installed and are no longer required:
  lxplug-volume python-colorzero
Use 'sudo apt autoremove' to remove them.
The following packages have been kept back:
  nodejs
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.`

linknum23, 2022-06-02:
That's interesting. Maybe our better ALSA configuration is saving us here.

Just a heads up, after some morning housekeeping the meat of this issue got merged in with #335