micro-nova/AmpliPi #244: Optical input

jackrosenthal, 2022-01-02:
I don't see an option in the WebUI for the optical input. What's it take to get this input working?

linknum23, 2022-01-02:
You need to do two things that aren't available in the web ui yet. Sorry about that. 

Create a FilePlayer stream using the api's create_stream with the json:
`{ 
 ​   "name":  "Input Playback", 
 ​   "type": ​"fileplayer"​, 
 ​  ​ "url": ​"alsa://plughw:cmedia8chint,0"​
 ​} 
 `

And ssh into the box and run the following ALSA command to switch the mux to the correct input:
`amixer -c 1 set 'PCM Capture Source',0 'IEC958 In'`

If this command fails, the USB Audio might be on a different card #: try 0, 1, 2, 3 until the command succeeds.

Line tells ALSA to set the mux to the optical input, if you want to use the aux input instead use 'Line'

Lohrer, 2022-01-06:
You can use `-D usb71` to avoid guessing the card number:
```
amixer -D usb71 set "PCM Capture Source" "IEC958 In"
```

closed 2022-01-18 by linknum23