micro-nova/AmpliPi #247: Only one Spotify input is seen

tynman, 2022-01-03:
I've created 4 Spotify inputs, and for the first day or so we were able to have multiple devices streaming to different areas of the house. Currently only one input can be seen, no matter how I switch the devices to different input channels.

I've tried refreshing the app page, rebooting the AmpliPi, and re-imaging to the latest update. No resolution to the issue so far. Any help you can give would be appreciated.

linknum23, 2022-01-04:
Do you mind sharing your config?  Go to **⚙️** -> **Configuration and Reset** -> **Download Current Configuration** and click **Download**

Otherwise a couple of things to check:

- Do each of the sources 1-4 have a different Spotify stream configured as their input? (Each of them need to be named differently)

- We have noticed sometimes a Spotify stream can get into a bad state, we are working on detecting this automatically
in the meantime, to fix a missing Spotify "device" go to the corresponding source tab (1-4) and select any other input like **Input 1 - rca**, then reselect the previously selected Spotify stream (rebooting AmpliPi will do something equivalent)

tynman, 2022-01-04:
I've attached the [config.json.txt](https://github.com/micro-nova/AmpliPi/files/7809045/config.json.txt) file for reference.

Inputs 1 and 4 didn't have anything assigned to them for a few days, and after assigning them to Tyler and Guest, respectively, they're now showing up in the Spotify app.

It's still not seeing the Kids Spot, which is assigned to _Input 3_. That's been assigned for several days, so it's "unchanged" from when this bug was opened. I've tried assigning Input 3 to RCA for several seconds, then back to Kids Spot, but that didn't resolve the issue.

Thanks for your help

linknum23, 2022-01-04:
Hmm that is wacky. I wish I could easily get the output of the underlying vollibrespot deamon to debug this a little further, but I would need you to ssh into your AmpliPi.

Can you try changing the name of `Kids Spot` to something else maybe just 'Kids'?

tynman, 2022-01-06:
I'll give that a try. Just by way of update, I think it can be reproduced by having the Spotify zones assigned, then having an input "cannibalize" one of the zones. E.g.:

Input 1 -> Spot A
Input 2 -> Spot B
Input 3 -> Spot C
Input 4 -> Spot D

Then change Input 2 to Spot C. Input 3 will now be without a source, and neither Spot B nor Spot C will show up in the Spotify app.

linknum23, 2022-01-07:
Just to clarify here. At the moment when a Spotify or other streaming input is 'stolen' from another source, that Spotify stream is stopped, then restarted and will need to be reconnected with the Spotify app. The source it was stolen from will now have nothing playing on it. The Spotify stream that was switched out is also stopped as it has no output to play to.

tynman, 2022-01-07:
@linknum23 yes, the behavior you describe is what I would expect.

As an update, the steps to reproduce that I gave you are not reproducing it. Currently only my "Spot Guest" source isn't showing up. In the _config.json_ it shows up as "state":"disconnected" as opposed to the other sources that show "state":"stopped". I've tried changing it to _stopped_ manually in the JSON and uploading that, but it reverts to _disconnected_.

Will keep updated if I'm able to find consistent steps to reproduce.