micro-nova/AmpliPi #248: Spotify Currently Playing Not Updating

bs42, 2022-01-04:
Sometime last night the currently playing info got stuck on my Spotify input.  New music has been streaming in just fine, but the album info, as seen on the amplipi.local interface as well as via the REST API hasn't changed in hours.

Is there any debugging information you'd like me to get from the unit before I reboot it to attempt to clear the issue?

linknum23, 2022-01-04:
Hmm looks like the metadata reader must have encountered some issue. Right now this is only debuggable by adding a verbose flag to the underlying metadata reader, which we used during development. We really need to make it easier to debug these types of issues.

To fix this issue there is no need to restart AmpliPi:
- Go to the corresponding source tab (1-4) and select any other input like Input 1 - rca, wait 5 seconds, then reselect the previously selected Spotify stream

bs42, 2022-01-04:
So now there's a new issue.  I did the above and the album info is now updating, however the button in the interface shows it's not playing and checking the REST API shows: "state":"stopped".

I fixed that by repeating the above steps but also going to the Spotify client and pausing and then resuming after reselecting the stream.

bs42, 2022-01-07:
Just to followup, this did happen again yesterday afternoon.  If there's anything I can do to help diagnose, I'd be glad to help.

linknum23, 2022-01-08:
Thanks for the report. We know that our Spotify integration needs some work, it's definitely been a pain point for several of you. 

It looks like there is a couple of things that we need to add to improve on streams I'm general:

- A stream watchdog/restart mechanism ( right now if the background program running the stream fails it stays in a failed state)

- A decent interface for viewing debug messages from the stream in the web interface. This would make for much more helpful bug reports.

The combination of these things would help us work out the quirks in the underlying Spotify implementation.

linknum23, 2022-01-11:
> I did have an issue this morning where 2 of my speaker zones showed muted when they were not.  I don't know if it's related but thought I'd mention it.  I "muted" and unmuted and that got the interface back in sync.

Sounds like the interface got out of sync somehow.

linknum23, 2023-02-01:
Closing this for now, since we have a new spotifyd integration.

closed 2023-02-01 by linknum23