micro-nova/AmpliPi #249: Feature: Bandcamp support

ibeardslee, 2022-01-10:
Bandcamp (https://bandcamp.com) allows streaming of music.

Kodi has an addon for Bandcamp that could be used as example code to bring into AmpliPi .. https://kodi.tv/addons/matrix/plugin.audio.kxmxpxtx.bandcamp

linknum23, 2022-01-11:
It looks like that kodi plugin uses https://github.com/Virusmater/bandcamp_api to interface with bandcamp's API. That would give an integration a decent start.

At the moment AmpliPi doesn't really have anything that looks like a real time song selecting interface, so an initial AmpliPi bandcamp integration would have to play something equivalent to a Pandora station where the stream is preconfigured in AmpliPi's settings. 

It looks like Bandcamp has something sort of equivalent that they call a song collection. Would a stream that played a song collection (or something similar) be useful?

ibeardslee, 2022-01-11:
Absolutely.  Bandcamp seems to consider all albums as part of a user's 'collection', only having a single collection.  This in itself is not completely optimal, but the way Bandcamp seems to do things.  My (personal) usage of Bandcamp is to play an album at a time, not sure if that would be doable or just another level of 'song selection'.

From my point of view, getting it to the level of being able to play an album would be 'perfect' I think just getting it into AmpliPi as best as can be done now would be grand.  Reading your "At the moment" comment regarding realtime song selection, as AmpliPi grows more features we can look at how we can expand the features of the Bandcamp support with those features.

linknum23, 2022-01-11:
We will keep that all in mind. Looks like we will need to know a little more about bandcamp and its api to figure out the scope of this and what makes sense. 

Thanks for looking into all of this!

ibeardslee, 2022-01-14:
Noting that there is also a Mopidy extension for Bandcamp https://pypi.org/project/Mopidy-Bandcamp/ which could make things a bit easier if it fits in with #66

linknum23, 2022-01-14:
At the very least, that would be a good second reference implementation. We still haven't looked into the best way to integrate Mopidy.