micro-nova/AmpliPi #260: Feature: Integrate Google Cast

linknum23, 2022-01-18:
We would love to integrate with the google cast ecosystem as a castable device.
Basically we would want to make AmpliPi look like one or more chromecast audios. 
From a basic search it appears this will require some coordination with Google as there are not any open source implementations.

imle, 2022-09-15:
I did some research here and it looks like Google has no intention of open sourcing their Google Cast receiver.

The original Chromecast Audio used a different and already reverse engineered casting protocol. I'm not sure how long that will remain supported despite Google saying they will continue to do so.

linknum23, 2022-09-15:
> I did some research here and it looks like Google has no intention of open sourcing their Google Cast receiver.
Realistically if we ask google nicely they may give us a binary that we can use in AmpliPi system. I would not anticipate them open sourcing the protocol.
> The original Chromecast Audio used a different and already reverse engineered casting protocol. I'm not sure how long that will remain supported despite Google saying they will continue to do so.
The last time we looked at that it was still not feasible to use in AmpliPi. IIRC the remaining issue on using that protocol is getting a valid key every 24 hours.

imle, 2022-11-27:
> Realistically if we ask google nicely they may give us a binary that we can use in AmpliPi system.

I'm not sure where to even reach out there. In my research it looked like you'd need a brand representative to get that level of access.

> The last time we looked at that it was still not feasible to use in AmpliPi. IIRC the remaining issue on using that protocol is getting a valid key every 24 hours.

I did not find a lot of documentation on the old protocol as most combinations of words I searched led to the new one. Do you recall what the issue with getting the key?

linknum23, 2022-11-29:
I haven't look into this in a while. When I did what it boiled down to was that a Chromecast device does a daily key request/update with Google's Chromecast API. That API requires an up to date key.

Sadly no efforts from the open source community were able to replicate the key request procedure likely due to it being encrypted in some way.