micro-nova/AmpliPi #262: Feature: Audio Pipes

linknum23, 2022-01-18:
Many dynamic features are not currently possible due to our basic ALSA integration.

One use case would be having several long running Spotify instances that when played to would output audio on a specific set of zones and when paused/stopped would not use an existing input/source.

Another use case would be adding some ducting to announcements (for the digital outputs at least).

- [ ] Investigate dynamic endpoint switching with ALSA using audio buffers
- [ ] Add event interface to api
- [ ] Add background capability to some streams

linknum23, 2022-10-24:
We did some research into this recently and have a working demo of a single stream.

Here's what we did:
Enable 2 loopback devices with: `sudo modprobe aloop`

Now we need to connect something to each side of the loopback:
Input Side: alsaloop -C plughw:2,1,4 -P ch0 -t 100000
Output Side: cvlc -A alsa --alsa-audio-device plughw:2,0,4 https://somafm.com/groovesalad64.pls

The loopback card creates two loopback devices with 8 channels a piece. They can be used bidirectionally which either gives us 8 or 16 total virtual outputs. depending on if these are mono or stereo channels.

If we need more we may need to configure the loopback module and build a similar kernel module.

Link that got us started with this: https://sysplay.in/blog/linux/2019/06/playing-with-alsa-loopback-devices/

linknum23, 2022-12-16:
To make this simple we added 4 loopback devices by putting the following in `/etc/modprobe.d/sound.conf`:

    options snd_aloop enable=1,1,1,1 index=2,3,4,5 pcm_substreams=2,2,2,2

These appear in the output of `aplay -l`:
```
**** List of PLAYBACK Hardware Devices ****
card 0: sndrpihifiberry [snd_rpi_hifiberry_dac], device 0: HifiBerry DAC HiFi pcm5102a-hifi-0 [HifiBerry DAC HiFi pcm5102a-hifi-0]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 1: cmedia8chint [ICUSBAUDIO7D], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
card 2: Loopback [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 1/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 2: Loopback [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 3: Loopback_1 [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 3: Loopback_1 [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 4: Loopback_2 [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 1/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 4: Loopback_2 [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 5: Loopback_3 [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 5: Loopback_3 [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 6: Loopback_4 [Loopback], device 0: Loopback PCM [Loopback PCM]
  Subdevices: 1/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
card 6: Loopback_4 [Loopback], device 1: Loopback PCM [Loopback PCM]
  Subdevices: 2/2
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
```

In the future there should be a way to make aliases for these devices as well.

linknum23, 2022-12-16:
We need to find a workaround for an issue we found in testing: 
- Playing audio out ch1-3 is causing a kernel fault
Here is an example failure:
`alsaloop -C plughw:2,1 -P ch1 -t 10000000 -v`
```
!!!Scheduler set to Round Robin with priority 99 FAILED!
Plug PCM: Linear Integer <-> Linear Float conversion PCM (S16_LE)
Its setup is:
  stream       : PLAYBACK
  access       : RW_INTERLEAVED
  format       : FLOAT_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 32
  buffer_size  : 4096
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 2147483647
  stop_threshold   : 4096
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Slave: Route conversion PCM (sformat=S16_LE)
  Transformation table:
    0 <- none
    1 <- none
    2 <- none
    3 <- none
    4 <- none
    5 <- none
    6 <- 0*0.64
    7 <- 1*0.64
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 2
  rate         : 48000
  exact rate   : 48000 (48000/1)
  msbits       : 16
  buffer_size  : 4096
  period_size  : 1024
  period_time  : 21333
  tstamp_mode  : NONE
  tstamp_type  : MONOTONIC
  period_step  : 1
  avail_min    : 1024
  period_event : 0
  start_threshold  : 2147483647
  stop_threshold   : 4096
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
Slave: Direct Stream Mixing PCM
Its setup is:
  stream       : PLAYBACK
  access       : MMAP_INTERLEAVED
  format       : S16_LE
  subformat    : STD
  channels     : 8
  rate         : 48000
  exact rate   : 48000 (48000/1)
[... 185 lines cut ...]
  period_event : 0
  start_threshold  : 2147483647
  stop_threshold   : 262144
  silence_threshold: 0
  silence_size : 0
  boundary     : 1073741824
  appl_ptr     : 0
  hw_ptr       : 0
Latency 441000 frames, 10000000.000us, 10000.000000ms (0.1000Hz)
playback ch1/capture plughw:2,1 sync type: CAPTRATESHIFT
New pitch for playback ch1/capture plughw:2,1: 1.00000000 (min/max samples = 0/0)
New pitch for playback ch1/capture plughw:2,1: 0.99998866 (min/max samples = -60/0)
Bus error
```

In our journalctl logs, Bus error corresponds to this kernel error:
```
Dec 16 16:41:19 amplipi kernel: Alignment trap: not handling instruction ed217a01 at [<76cdd424>]
Dec 16 16:41:19 amplipi kernel: 8<--- cut here ---
Dec 16 16:41:19 amplipi kernel: Unhandled fault: alignment exception (0x801) at 0x74c3c13e
Dec 16 16:41:19 amplipi kernel: pgd = 849a3a81
Dec 16 16:41:19 amplipi kernel: [74c3c13e] *pgd=05273835, *pte=220d475f, *ppte=220d4c7f
```

There is likely a bug in the **aloop** kernel module that we will need to track down.

linknum23, 2022-12-16:
Next steps: 
- [x] Update to the latest pi kernel
- [ ] Update to newer? aloop kernel module
- [ ] Add more debugging info to aloop kernel module?

linknum23, 2022-12-20:
This works in 64-bit. It may be our needed push to make #335 a reality .

linknum23, 2023-01-06:
I have been testing with 2 running loopbacks but  I'm having an issue that I didn't notice initially on the USB DAC outputs. Every ~5 seconds I'm hearing a clicking sound.  I've explored most of the parameter space of alsaloop and haven't had any luck reducing the amount of clicking.

Looking at the performance in HTOP it looks like a stream flowing through aloop uses about 10% of one CPU. This is small but not insignificant. We will need to test with more running streams to see how much we can push this.

linknum23, 2023-01-10:
Alright it looks like the problem all along was that we were using plughw on the capture side of alsaloop.

Here's the combination of commands that work well across all audio outputs: 
- ` alsaloop -C hw:Loopback_1,1 -P ch1 -t 100000`
- `cvlc -A alsa --alsa-audio-device dmix:CARD=Loopback_1,DEV=0 https://somafm.com/groovesalad64.pls`

This combination also works in 32-bit, so we won't be forced to upgrade to 64-bit just yet.

linknum23, 2023-01-12:
Testing for this feature is in the [use-loopback](/micro-nova/AmpliPi/tree/use-loopback) branch.

linknum23, 2023-01-17:
A second version of this is being worked on in [virt-streams](https://github.com/micro-nova/AmpliPi/compare/develop...virt-streams). 

The idea in this branch is to "just" add support for long running backlground streams without any need for fancier virtual sources.

linknum23, 2023-03-15:
This development work is paused waiting for UI changes to make it useful

rtertiaer, 2024-01-08:
@linknum23 do you feel like this is complete now?

linknum23, 2024-01-08:
Here's what I see that is remaining and related, at this point these should just be separate issues though

- [ ] Add a list of long running streams that have the options to add/remove/reset each stream
- [ ] Add event handling for IFFT-style automated switching (hopefully simpler than IFFT tho)
- [ ] Handle over-allocation, one long running stream needs to be kicked out, which one?

closed 2024-01-08 by linknum23