micro-nova/AmpliPi #283: No playing, UI behaves weirdly

0lek, 2022-02-04:
Hey,

I just received the unit last week and finally installed it yesterday. I tried playing internet radios or spotify, but there's no sound coming out, but I don't think it's a hardware issue:

When I select the radio stream, I see the controls next to the logo, but: 1) There's no play button - all the others yes, no play button and 2) the buttons disappear within a second or so.
Something similar happens when I select spotify: I see the logo for a second and then the logo disappears and there is the average char displayed (as in cross through circle).

It would seem that maybe there's a network problem, but adding a new internet radio pulls the correct logo, so it should work.

Have no hand analogue source to check.

How can I troubleshoot? 

Aleks

linknum23, 2022-02-04:
I'm sorry you are having trouble with this.

First off let's make sure some audio works:

Your AmpliPi shipped with the internet radio station groove salad. Try
selecting that and making sure that you have added the zones you are trying
to play to.

After selecting groove salad you should see some song info appear. Note
that the play, next, prev controls will quickly go away because internet
radio doesn't currently support playback controls.

Let me know if that works and we can continue from there.

0lek, 2022-02-04:
Hey. I tried the groove salad station as well. Same thing as with my other streams.

linknum23, 2022-02-05:
Are the zones you are trying to play to "connected" to the same audio
source?

0lek, 2022-02-05:
Don't understand your question?

I have one zone (1) connected - all the other zones are disabled. Below you see screenshots. The "none" shows for like 0.5s - then it disappears.

It *might* be some kind of internet issue? Is there a way to ssh into the amplipi and ping? That would be the best way to see whether it has network...

[image]
[image]

0lek, 2022-02-05:
Ok, it works. I Put the amplipi on another VLAN and it works. I have no idea why it didn't work, but works now. :) Still don't understand how to make Spotify work but will figure it out eventually ;)

Thanks

linknum23, 2022-02-06:
Glad to hear it.

I think you are the second person to have VLAN problems. Sorry again. We
will need to add some way of indicating the AmpliPi is offline, especially
for streams that require internet access.

0lek, 2022-02-06:
Hey,

The exact problem - I think - was that the VLAN did not specify a DNS server in the DHCP lease. Once I put it into the other VLAN, confirmed to work, I put it back on the original one, added DNS and it works. I still don't understand how it pulled the cover images for the streams without a DNS, but let's just assume it was magic.
What does NOT work is spotify, but that is probably because Spotify connect uses shitty multicast and has no option of direct IP connection - I SO HATE this. 

To your suggestion with the offline:
I would want to have a way to ssh into the amplipi, because then I can simply verify / connect.
Oh and also please add an option to restart in the Web GUI.

linknum23, 2022-02-06:
You can ssh into AmpliPi at anytime. It is a raspberry pi with the default
'pi' user and 'raspberry' password and ssh enabled.

Please note that future units will ship with a randomly generated password
and some better security settings but we wanted the initial units to be as
easy to access as possible.

bs42, 2022-02-07:
> What does NOT work is spotify, but that is probably because Spotify connect uses shitty multicast and has no option of direct IP connection - I SO HATE this.
> 

I found the only way I could get Spotify Connect to work was to put my phone into the same VLAN as my AmpliPi so that it could find it in the Spotify app, then it would show up elsewhere (i.e. https://open.spotify.com) as a target device.

That said, I can't deal with Spotify's complete inability to shuffle my 1k+ item playlist so I have Mopidy running on an Intel NUC outputting audio into one of the RCA inputs.  It pulls in all my Spotify+local music and shuffles them properly.  As a bonus, no more need for Spotify Connect.

Hoping to figure out how to install Mopidy onto the AmpliPi one of these days.

linknum23, 2022-02-07:
You should also be able to run an mDNS repeater to bridge mDNS between subnets. The issue with Spotify and similar things like Airplay is that they use mDNS which by design only works on a single subnet.

0lek, 2022-02-07:
> You should also be able to run an mDNS repeater to bridge mDNS between subnets. The issue with Spotify and similar things like Airplay is that they use mDNS which by design only works on a single subnet.

Haha. I know that. Issue is that theory and practice are two things. I haven't yet made it work on pfsense. Will get there eventually.

0lek, 2022-02-07:
So I got it to work. I *think*. Still not 100% sure whether it works and will work or whether somehow spotify remembers the AmpliPi and will forget at some point. I assume it works. So my case is pfsense 2.5.2, unifi APs, two different VLANs. What I had to do to get it working:

1) Set up avahi in pfsense
2) Have firewall rules permitting MDNS / SSDP
3) Have a rule passing connections between the iphone VLAN and the AmpliPi VLAN on ephemeral ports. Now the kicker is: The IANA definition of 49152–65535 did not work. It only started working after I changed this to 32768-65535. This is so super-specific, it might actually help somebody someday, so I just leave it here.

AWESOME product BTW guys!

linknum23, 2022-02-14:
Thanks again for figuring all of this out. Closing this out it looks like we need to add an offline indicator for our web ui and some good documentation for advanced networking and VLAN setups.

- Added a note in #34 
- Added #293

closed 2022-02-14 by linknum23