micro-nova/AmpliPi #287: Feature: Add Coral TPU for better Text To Speech based announcements

linknum23, 2022-02-09:
Text to Speech would be a really nice integration for announcements. Right now it needs to happen externally to the announcement. Figuring out a simple way to integrate it into announcements would make them a lot more useful. 

For supporting this in conjunction with something like the coral we would need to do the following: 

- [ ] Find TTS Neural net
- [ ] Test with USB attached coral
- [ ] Add TTS field in announcement that is supported if attached

@Lohrer and @klay2000 thanks for the inspiration on this

Lohrer, 2022-02-09:
Thanks really belong to @sepffuzzball in providing some possible uses for Home Assistant controlling AmpliPi https://github.com/micro-nova/AmpliPi/discussions/206#discussioncomment-2140029.

TTS anouncements would be awesome!

linknum23, 2022-02-09:
One potential network: https://github.com/mozilla/DeepSpeech. I wonder how much CPU it uses without a dedicated TPU?

Lohrer, 2022-02-09:
A possible TTS implementation would be https://github.com/coqui-ai/TTS. I tried installing with pip on a Pi but it failed, looks like we'll have to do a manual install due to some of the dependencies (at least torchaudio) not providing arm binaries.