micro-nova/AmpliPi #292: Feature Enhancement Request: Per Zone Volume for Announcements

bs42, 2022-02-12:
I recently started to use the Announcements feature (love it BTW) and find that I would really like the ability to have the volume configured per zone.

### Use Case:
I have automations that change the volume in the house based on time of day, so at night I have the volume low, daytime higher, etc.
This precludes the option to use the "current" volume for any high priority type announcement such as "Exterior Security Door Opened!" as I don't want that to be a whisper obviously.

Then there are the pre-out zones, those need to be 100% volume all the time because their volume is controlled by the device they are connected to.

### Implementation idea:
Since no one likes breakage, I was looking over the existing methods and I think potentially the easiest method to implement this request would be to leverage the "presets", (not knowing the backend of this I could be wrong), but the idea being that `/api/announce` would add an option along side `zones` or `groups` called `preset` and it would exclude `vol`,`zones`,`groups` if provided and would be able to have a list of zones and their volumes for this announcement call, and then restore back to the previous values when done.

linknum23, 2022-02-13:
Adding a preset option should be very easy to implement for announcements especially since behind the scenes announcements actually create a temporary preset for the specific purpose of setting volumes. The preset option would just replace that.

linknum23, 2022-05-31:
Something similar is possible in 0.1.8 by using the `vol_f` (instead of the `vol`) field and changing the max volume of each of the zones max volumes using the `vol_max` field. For your use case I would recommend having a preset for each time of day that configured each of the zone's max volumes and add automation to execute the preset on each time transition. 

Here's what a couple of the presets would look like in Jason's (simplified) house:

**Sleeptime**

```json
   {
      "name": "Sleeptime",
      "state": {
        "zones": [
          { "id": 0, "vol_max": -40 },
[
          { "id": 0, "vol_max": -40 },
[
          { "id": 0, "vol_max": -40 },
          {
            "mute": true,
            "id": 1
          },
       
```

**Office Hours**

```json
```

An announcement using vol_f would look like:

```json
```

bs42, 2022-05-31:
I'll have to give that a try.

One minor nit though, I notice it's not possible to set the min vol to 100%, do you think we could get that in an update?  For the line-out I need 100% or else it's almost inaudible.

linknum23, 2022-06-01:
We added an issue for this: #328