micro-nova/AmpliPi #293: Add documentation for using on a VLAN

linknum23, 2022-02-14:
Many more advanced users have (or want to have) restricted VLANs for IOT devices such as AmpliPi for security and privacy reasons. We need to test and document how to configure a network to use AmpliPi on a restricted VLAN setup.

linknum23, 2022-02-14:
The end result of this would probably be the creation of `docs/ANETWORKING.md` with some documentation on what needs to be done and an example setup with some off the shelf networking hardware.

brianhealey, 2022-06-01:
Hmm..  I imagine this would be a manual (ssh) process considering the box requires DHCP to start.. 

I use ubiquiti switch hardware that applies vlan on the switch interface vs the device and that works fine without requiring hardware/os changes.