micro-nova/AmpliPi #298: Add AmpliPi interface to Sonoff Touch Panel

linknum23, 2022-02-23:
- [ ] Checkout development process
  - [ ] UI Changes
  - [ ] ESP32 FW Changes
- [ ] Add basic status polling interface to amplipi (kjk2010 did this here https://github.com/kjk2010/AmpliPi-POE-Touchscreen)
- [ ] Add song info display
- [ ] Add album art? (This may not be possible)
- [ ] Add volume control

linknum23, 2022-02-28:
Lots of info can be found in this thread: https://community.home-assistant.io/t/sonoff-nspanel-smart-scene-wall-switch-by-itead-coming-soon-on-kickstarter/332962/48

brianhealey, 2022-03-05:
Are you planning on using espHome or directly interfacing with the ampliPi?

linknum23, 2022-03-05:
For the initial port we are interfacing with it directly (using
micropython) but we are open to integrating with esphome as well.

yoggiland, 2022-03-12:
Hi,
I understand that there are two versions, a US and an EU. From what I read they are different do to screen rotation etc. see link.

https://community.openhab.org/t/nxpanel-replacement-firmware-for-sonoff-nspanel/132869/72
https://community.openhab.org/t/nxpanel-replacement-firmware-for-sonoff-nspanel/132869

I am guessing you have the US version, do you think your current approach (using micropython) will work with the EU version?

yoggi

linknum23, 2022-03-12:
We don't have an EU version on hand yet but it looks like it UI will just need a couple of layout changes from the US version. The EU and US versions share the same hardware, just with different power inputs and display resolutions. The backend running on micropython will hopefully be the exact same.

rickballard, 2022-03-25:
When do you guys expect to be able to sell some of these wonderful wall keypads, to eager customers, like myself?  Can you be persuaded to sell off some of your prototypes for testing...?

linknum23, 2022-03-25:
Based on where we are right now (a working prototype). I would guess we
will have the code available in a month for those that want to load the
code on the devices themselves.

As for the having them on our website with a finalized AmpliPi touch
interface, I would guess they would be available in 3-6 months.

rickballard, 2022-03-27:
You do realize that we are all loving you guys up here, like a feast for starving piglets...?  :)

linknum23, 2022-05-31:
We are getting close to being able to sell [these](https://www.micro-nova.com/amplipi/store/amplipi-in-wall-wifi-touchscreen-controller-with-integrated-2-device-switch) and will be making the related repo public in the next couple of days!

brianhealey, 2022-06-01:
Let me know if you need a tester.  I have the US version in hand and can flash it easily.

rickballard, 2022-06-01:
Yes, please, a tester would be wonderful, thank you.  It looks excellent.  

I particularly like the WiFi-ness, as legacy Cat \-wiring tends to make folks think of the keypad as being zone-specific, (even though this keypad can control all zones from a single location).  You might sell more keypads if they can be set to display a default zone dashboard, then folks might buy one per zone instead.  

Also, those hard-buttons are just itching to have a default smart-home purpose in life.  Please consider this idea:

Post simple instructions, so folks can add the circuit out of a cheap Lutron On-Off Pico, to be operated by those two (spare) hard-buttons.  This could add $100 in value to those folks who already have smart-lighting, at the cost of a few extra bucks for the Pico and 5-minutes of soldering/reassembly.  Alternatively, you guys could more formally ask Lutron to let the keypad talk to their hub via WiFi, and then there would be no hacking necessary.  

Those buttons are just itching to have a default smart-home purpose in life.

linknum23, 2022-06-01:
These touch panels are actually available for sale, now!  Feel free to purchase one and let us know how it works.

As for the hard buttons: at the moment the hard buttons are able to control two led light circuits using twon onboard relays. The intent is that this could replace a single or dual light switch in your house simply by removing the switch and wiring in this wall panel in place.
We are hoping to expose them as smart switches in the future so they can be controlled by something like home assistant.

> Also, those hard-buttons are just itching to have a default smart-home purpose in life. Please consider this idea:

I'm not sure if connecting the output of one of the wallpanel's relays to a pico will do what you want. Hopefully we can add the Smart Home use case soon.

linknum23, 2022-06-01:
Does anyone know of a simple MQTT light interface that we can provide for the 2 relays?

brianhealey, 2022-06-01:
I use homeassistant with the mqtt addon to run most of my tasmota devices. would that work?

rickballard, 2022-06-02:
OK, purchased, will let you know how it works up here in Canada.  I plan to turn it into a weather-resistant remote, that rests in a dock on an outside wall.  I can hide it inside in the winter, so the Polar Bears don't eat it, but it may have to cope with some hot and humid weather this summer.  Thanks.

linknum23, 2022-06-02:
Looking at the panel specifications it is definitely intended for indoor use.

[image]

rickballard, 2022-07-03:
Touch panel works as advertised!  However, have not yet figured out how to 
a) link the ewelink app to the panel so that the hard buttons can be set to operate other sonoff/ewelink-compatible devices, like other separate mains-power relays.
b) link the panel to the phone via bluetooth, because the bluetooth discover phase of the power-up cycle seems to have been bypassed by the AmpliPi software load. 
c) Meanwhile, I noticed that if the Amplipi app is running on my phone, the panels seem to get ignored, or they work so slowly that I have not yet been patient enough to wait for them to respond.

linknum23, 2022-08-02:
Closing this since the panel's repo is public now. Check it out at micro-nova/AmpliPiWallPanel/

closed 2022-08-02 by linknum23