micro-nova/AmpliPi #304: Fix power supply override and document power register

Lohrer, 2022-03-30:
The 9V and 12V power supplies have an enable input that preamp_i2c_regs.md says is controllable but the firmware will in fact always immediately turn the supplies back on. There's not much reason to ever turn off the 9V or 12V, so alternatively maybe these bits should be read-only.

Once a decision is made over control, create better documentation and update any existing documentation.

Lohrer, 2022-06-03:
The 9V enable is only present on old hardware, and the 12V fan power should always remain on. This register should be changed to be read-only.

linknum23, 2023-02-02:
@Lohrer did this get fixed?

Lohrer, 2023-03-01:
The work was done in #340, and the last bit of documentation was fixed in #452

closed 2023-03-01 by Lohrer