micro-nova/AmpliPi #307: Fix preamp checkout test

Lohrer, 2022-03-30:
I don't have time to debug this now so I'm leaving this here. Running `hw/tests/built_in.bash preamp` results in:

```
configuring amplipi for testing
waiting for config file to be written
Running test 'preamp'. Press Ctrl-C to stop.
Test will play Analog 1 Left, Analog 1 Right...Analog 4 Right, Digital 1 Left... Digital 4 Right
- Verify that each side and all 8 sources are played out of each of the 6 zones
Traceback (most recent call last):
  File "/usr/lib/python3.7/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.7/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/amplipi-dev/amplipi/tests.py", line 309, in <module>
    preamp_test(ap, exp_unit=args.expansion)
  File "/home/pi/amplipi-dev/amplipi/tests.py", line 252, in preamp_test
    digital_msgs = [models.Announcement(source_id=src, media=f'web/static/audio/digital{src+1}.mp3', vol=-30) for src in range(4)]
  File "/home/pi/amplipi-dev/amplipi/tests.py", line 252, in <listcomp>
    digital_msgs = [models.Announcement(source_id=src, media=f'web/static/audio/digital{src+1}.mp3', vol=-30) for src in range(4)]
  File "/home/pi/amplipi-dev/venv/lib/python3.7/site-packages/pydantic/main.py", line 406, in __init__
    raise validation_error
pydantic.error_wrappers.ValidationError: 1 validation error for Announcement
vol
  ensure this value is greater than or equal to 0.0 (type=value_error.number.not_ge; limit_value=0.0)
```

linknum23, 2022-03-30:
Looks like we broke a small part of the api

linknum23, 2022-03-31:
> Assigned #307 <https://github.com/micro-nova/AmpliPi/issues/307> to
> @linknum23 <https://github.com/linknum23>.

Looks like the announcement model needs to have a vol/vol_f like the rest
of the code.

closed 2022-04-25 by linknum23