micro-nova/AmpliPi #311: Data-whistling noise

rickballard, 2022-04-04:
When selecting an unused rca input, the speakers will issue a high-pitched, but very quiet, whistle, with occasional-buzzes in the background, much like very-fast-modem data noise.  

This noise can be almost-imperceptibly altered in pitch, by changing output settings in the web app then pressing save. 

 Is this a software bug, or is it just IF-audio-out inductive coupling on the board?  

Can the whistling noise be filtered out any more effectively, as it can cause one to gradually go mad if the speakers are overhead in one's otherwise silent office?

kjkoning, 2022-04-04:
I also have this issue, although mine happens with both used and unused RCA inputs. It also happens with the unlabeled dropdown option on the web interface stream selection.

It seemed to get louder with the 0.1.7 update, but I can't confirm that. Initially I thought it might be the audio output from the sound card interfaces, so I played with the volume levels in the alsa mixer, but that didn't seem to help, although I might have been modifying the wrong volume settings.

The issue goes away when muting the zones, so I have to make sure the bedroom zone is muted every night, otherwise it drives me crazy.

Lohrer, 2022-04-05:
Are either of you guys powering an audio device via the rear USB ports on AmpliPi? There is a known issue where plugging in say a Chromecast Audio to the USB port and its audio output to the RCA inputs causes extra noise. The USB ports use the power supply for the digital components while all the audio inputs use a separate power supply for the analog components. Connecting the two grounds through an external device tends to cause all the electrical noise the Pi is generating to get into the audio signals.

rickballard, 2022-04-05:
I am using the USB FM zipstick thingy that you guys recommended, (SDR-something?).  It is plugged into the back of the AmpliPi, and its coax cable goes to a FM-antennae on the roof via a ChannelMaster 8-port TV-signal amplifier.  So, this too might inject some noise, but I kinda doubt it would also amplify the FM signal as that gets band-passed separately, I think.  Anyhow, could any of this be the cause the maddening whistling noise, (on all zones when a silent or rca input is selected)?  Hmmm, maybe I should go and disconnect that zipstick thingy to see if it helps with the whistle...?

linknum23, 2022-04-11:
It looks like there are a couple of things being discussed here:

1. Noise on unused RCA inputs. This is expected. In the future we are planning on adding a disabled setting for each of the RCA inputs to remove the input as an option. There is some possibility of auto-detection of open inputs in the future but disabling the input will be much simpler in the short-term.

1. Noise on connected RCA inputs. This is not expected, but as @Lohrer said could be caused by using AmpliPi's USB power to power an audio device like a chromecast fed back in to an RCA input.
> It seemed to get louder with the 0.1.7 update, but I can't confirm that. Initially I thought it might be the audio output from the sound card interfaces, so I played with the volume levels in the [...]

The RCA inputs are not routed through the PI so alsa changes will not have any effect.

rickballard, 2022-04-20:
Getting the same noise on off-air and internet radio inputs, as well as the unused RCA inputs.  I suspect there is some crosstalk somewhere....

rickballard, 2022-05-19:
Also get the hissing whistle on the Spotify input, before and after it is connected.  Will test further to see if it is that antennae thumbdrive a to d converter device.

jonathanslenders, 2022-05-24:
Not sure if my situation is related. I'm getting some noise on the speaker output as soon as the output is enabled in the web UI. There is no noise at all on the pre-out, and it happens regardless of the volume setting, whether or not something is playing or the selected input. (Of course, it's more noticeable if nothing is playing.) It looks like I'm getting the same noise on all speaker outputs. I don't have any USB devices connected.

gorski123, 2022-05-24:
Hey @jonathanslenders are you getting this when you select the analog RCA input or a digital streaming source (Pandora, Spotify, Airplay, etc)?

jonathanslenders, 2022-05-24:
@gorski123. For both. Also for streaming sources. It happens as soon as the output is enabled, even at lowest volume. I don't hear the same noise using an active speaker connected to the pre-out.

gorski123, 2022-05-24:
Thanks @jonathanslenders. That sounds like a hardware/wiring issue with your box. I'll shoot you and email and we can debug it.

rickballard, 2022-05-25:
I have not yet had time to look into this exact same issue for my AmpliPi, so please post any results of your hardware debug. 

Given that the noise is distinct, like a whistling buzz from a 1990's dial-up modem, could you post the sound so we can compare the noises from different customers' AmpliPi...?

rickballard, 2022-06-02:
So, my apologies for the delay in testing this.  The whistling modem-buzz (WMB) is coming from the FM-dongle.  When the dongle is not plugged in, the input is silent, but otherwise, the WMB is quite loud in the background on all inputs, including internet radio input (BBC World Service in this case).  I suspect there is some RF interference being picked, as the dongle has an amplified signal going into it and the cable to the root-top FM-antennae is about 50ft.  The Channel Master signal amplifier is in the attic, just below the antennae.  I could try bypassing this amp to see if this fixes the loudness, but I suspect surface effects and grounding may be creating some cross-talk, since the dongle is messing with all inputs.