micro-nova/AmpliPi #312: No Audio playing on Develop?

linknum23, 2022-04-06:
Need to investigate, looks like we broke alsa very recently

linknum23, 2022-04-19:
Looks like this was caused by something lingering in my config. Closing for now

linknum23, 2022-06-23:
It was a bad local ALSA config that caused the error I was seeing.  I have very rarely (twice) seen an issue where the USB DAC needs to be reset by a reboot, in this case one or more of sources 2-4 stop working until after a reboot. In this case source 1 is still working. It sounds like you may be running into a similar issue, to confirm I would try playing audio out the 1st source?

sumnerboy12, 2022-06-23:
Yeah source one was a radio station and that wasn't working. I am pretty
sure I tried every source with no joy.

I tried stopping and restarting the radio stream but no luck either.

It does sound similar but it was for all sources. I've had it happen a
couple of times since upgrading 0.1.8.

Will keep an eye out.