micro-nova/AmpliPi #323: All configuration and settings erased

datagitlies, 2022-05-05:
On Monday (5/2) I was trying to play some music and noticed my AmpliPi was not responding. I tried controlling the unit from the Android App and Google Chrome (on Android) with no luck. I could get the UI to load in the browser but was unable to start / stop any stream or adjust the volume of any zone. I could load the API spec (`/doc`) so I know the unit was responding to requests, just not any audio controls. So, typical impatient user... after a few minutes I figured I would power it off and back on again. When the AmpliPi powered back on, all of my streams / zones / groups / etc. were gone.  It's as if the device factory reset itself.  It won't be too difficult to add everything back but I wanted to post here to see if there was some sort of way to provide logs to you all to see what could have happened.  I'm not really worried about recovering my configuration and settings.  I simply want to make sure this doesn't happen to current / future customers of this amazing product!  Anything you'd like me to do (get logs, post here, etc.) I'm happy to do.  Let me know how I can help!

P.S.  When this occurred I wanted to check the issues of this project to see if others were experiencing anything similar.  In doing that I happened to notice that there was a new beta or alpha release of `0.1.8` just 3 hours prior to when my issue occurred.  Not sure if that had anything to do with it or what.  Does the AmpliPi proactively look for updates / download them?  Could I have power cycled during that process which caused my settings to be wiped?

linknum23, 2022-05-05:
Sorry you are having trouble with this, you may have hit a combination of several issues at the same time.

> my AmpliPi was not responding
Sounds like you encountered #218 that is fixed in the latest 0.1.8 release. Sorry about that!

>  It's as if the device factory reset itself
This is weird and would only happen if the config file and its backup both got corrupted somehow during the power off. This should not be possible but we will see if we can replicate it on our side. It may be related to the bad state the system was in from #218.

 > Does the AmpliPi proactively look for updates / download them?
No, We prefer the user to have full control of the update process. To update follow the instructions here: https://github.com/micro-nova/AmpliPi#updates-and-releases. That said we may add auto-updates as an option that you can opt in to in the future.

One suggestion I would make just in case this happens again (hopefully it won't) is to save the configuration after you have set your AmpliPi back up again.

1. Click the gear icon (gear) in the upper right corner to go to the configuration page
2. Select **Configuration and Reset** and go to **Download Current Configuration** 
3. Click the **Download** button

If you ever need to restore the configuration its a similar process:

1. Go to the configuration page
2. Select **Configuration and Reset** and go to **Upload a Configuration** 
3. Browse for the saved config file
4. Click the **Load** button

linknum23, 2022-05-05:
Our experience with #218 is you get into a degraded state before the system being unresponsive. Upgrading the kernel (what 0.1.8 does) fixed the issue in our testing. 

To send us a log:

    ssh pi@amplipi.local
    journalctl --user-unit amplipi > unresponsive-issue.log
    scp pi@amplipi.local:unresponsive-issue.log .

Then either add **unresponsive-issue.log** to this issue or send it to our support email: support@micro-nova.com.

After that feel free to update to 0.1.8 and tell us if you continue to experience similar issues (hopefully not).

datagitlies, 2022-05-05:
Yep, something was definitely wrong with my config file:

```
May 02 18:17:07 amplipi systemd[544]: Started Amplipi Home Audio System.
May 02 18:17:20 amplipi authbind[789]: error loading config file: Expecting value: line 1 column 1 (char 0)
May 02 18:17:20 amplipi authbind[789]: using default config
```

The full set of log events for Tuesday (5/2) are attached in [unresponsive-issue.log](https://github.com/micro-nova/AmpliPi/files/8633709/unresponsive-issue.log)

I'm surprised that the config file and its backup both got corrupted.  I can now see why taking a backup of my configuration every so often might be a good idea. Given what I see in [ctrl.py](https://github.com/micro-nova/AmpliPi/blob/main/amplipi/ctrl.py) both files would have been overwritten with [`DEFAULT_CONFIG`](https://github.com/micro-nova/AmpliPi/blob/main/amplipi/ctrl.py#L98) so no real hope of knowing what was wrong with the files at this point (or if they even contained any data) is that correct?

datagitlies, 2022-05-05:
Something that's noteworthy here... from what I can surmise from the code in [ctrl.py](https://github.com/micro-nova/AmpliPi/blob/main/amplipi/ctrl.py) and the logs, my backup config file was never even loaded. So the backup may have been in a good state and not actually corrupt (but then it was quickly overwritten 😢).  I'd be happy to contribute a tiny fix to `ctrl.py` that more gracefully handles this scenario.  Just let me know if that's something you're interested in or if you agree / disagree there is even a bug (I believe I see one - specifically in the exception handling).

linknum23, 2022-05-05:
Looking at the code I see one issue that would cause the log message you are seeing:
```python    
    # test open the config file, this will throw an exception if there are issues writing to the file
    with open(settings.config_file, 'a'): # use append more to make sure we have read and write permissions, but won't overrite the file
      pass
    self.config_file = settings.config_file
    self.backup_config_file = settings.config_file + '.bak'
    self.config_file_valid = True # initially we assume the config file is valid
    errors = []
    if config:
      self.status = config
      loaded_config = True
    else:
      # try to load the config file or its backup
      config_paths = [self.config_file, self.backup_config_file]
      loaded_config = False
      for cfg_path in config_paths:
        try:
          if os.path.exists(cfg_path):
            self.status = models.Status.parse_file(cfg_path)
            loaded_config = True
            break
          errors.append('config file "{}" does not exist'.format(cfg_path))
        except Exception as exc:
          self.config_file_valid = False # mark the config file as invalid so we don't try to back it up
          errors.append('error loading config file: {}'.format(exc))

    if not loaded_config:
      print(errors[0]) ## TODO: THIS ONLY PRINTS THE FIRST ERROR IF BOTH FILES ARE CORRUPTED 
      print('using default config')
      self.status = models.Status.parse_obj(self.DEFAULT_CONFIG)
      self.save()
```

linknum23, 2022-05-05:
Feel free to open a PR with a fix and a test to validate the fix. The test cases for loading the config can be found in `tests/test_ctrl.py`.

datagitlies, 2022-05-05:
I didn't see this:

```
## TODO: THIS ONLY PRINTS THE FIRST ERROR IF BOTH FILES ARE CORRUPTED
```

So maybe both my files were actually corrupted.  I'll play around with it and see if I can replicate what I think happened via a test.

linknum23, 2022-05-05:
Yeah I added that for illustration purposes. That line should be replaced with `for msg in errors: print(msg)` or something similar.

linknum23, 2022-05-09:
I was working on adding a reboot button settings interface on Friday. As it turns out the updater already has this functionality built it, we just need to make it easily accessible.

datagitlies, 2022-05-09:
Nice! Does the updater have _both_ reboot and shutdown functionality?  I would be most interested in shutdown since my use of it would be to gracefully shutdown the AmpliPi before I move it, change / add wired connections or add an expander unit.  Reboot is nice, but shutdown is what I'd be wanting.

As for this issue, go ahead and close it if you want.  I've updated to `v0.1.8` and have reestablished most of my configuration (plus saved a copy of my config).  So far no issues.  I highly doubt I'll run into this again especially since I have a workaround for a graceful shutdown (via ssh).

linknum23, 2022-05-09:
It will soon! Thanks for the suggestion.

closed 2022-05-09 by linknum23