micro-nova/AmpliPi #326: Add Notitications to Web Interface

linknum23, 2022-05-31:
We should add a small set of icons to provide useful notications via the user interface (adding similar information to the api as well)
- [ ] New release available (added to info->latest_release)
- [ ] Offiline indication
- [ ] Announcement in progress (info->announcement)

Additional info will be displayed with the presets drop down from the amplipi logo
- [ ] Show latest release with button to navigate to updater
- [ ] Show disconnected info (disconnected from amplipi or internet access unavailable?)
- [ ] Show announcement message?

SteveMicroNova, 2024-10-15:
This would best be implemented with an upgrade and more general adoption of the components in the `StatusBars` folder, those already support success, fail, warning, and info popups:
https://mui.com/material-ui/react-snackbar/

SteveMicroNova, 2024-10-16:
I've changed the tags and milestones so that this might end up in our modern triage meetings
It has the enhancement label so it won't end up on our high priority list, but perhaps it will end up in short order regardless

Having more documentation as to what should result in a notification would be appreciated