micro-nova/AmpliPi #328: Keep preouts at max volume

Lohrer, 2022-05-31:
The preout level is controlled by the volume controllers. Many applications have their own volume control and want the preouts at true line-level all the time. Currently there is a forced 20dB range from the min_vol and max_vol of a zone, so it's still possible to accidentally change the volume or have an undesired attenuation when playing an announcement. This needs a bit more discussion, but I can see two solutions:

1. Allow the max and min vol to both be the same. The volume sliders will have to be fixed to handle this situation.
2. Allow a zone to be marked as a 'pre-out' and/or 'line-level' only zone. If 'pre-out' only then software can take advantage of the AMP_ENABLE register in the preamp's firmware to avoid turning on the amps unnecessarily.  If 'line-level' only, the volume would stay at 'max_vol' always.

jonathanslenders, 2022-05-31:
Solution 2 would be great. There's a good chance I'll switch to an external amp for several zones (or all) and so it would be great if we can avoid having the built-in amp consume energy for nothing.

brianhealey, 2022-06-02:
This will be available still via the API, correct? I will need to work this into the HA integration and I would hate to have that logic leak externally.