micro-nova/AmpliPi #334: Changing a zone's source should unmute zone.

linknum23, 2022-06-01:


linknum23, 2022-06-01:
Yup, Also this would be in addition to #332 so the use case of moving a zone to a different source just to avoid playing to it will be handled there.

rickballard, 2022-07-05:
And an "everything off" button, or swipe-shortcut, would be nice.  This would avoid a wife wanting to whack the AmpliPiSonoff wall-contrller with a frying pan...

linknum23, 2022-07-05:
Planning on adding this to the wall panel already. Check out the related
issue here: https://github.com/micro-nova/AmpliPiWallPanel/issues/16

rickballard, 2022-07-05:
Hah, I do not see an entry for a "whack-off gesture", what are you calling
this new feature?
*Rick Ballard  416.414.8100 *

rickballard, 2022-07-05:
> This will use the existing Presets feature of AmpliPi which has one default
> preset called "Mute All". Here's an example of how it works in the app:
>
> https://github.com/micro-nova/AmpliPi/raw/main/docs/imgs/app_demos/load_preset_small.gif

On the Amplipi Sonoff wall controller, can we have a less than zero notch,
on the volume slider, which folks will instinctively know means "mute
all"...?

Regards, Rick.  416.414.8100