micro-nova/AmpliPi #335: Migrate to 64-bit Raspberry Pi OS

Lohrer, 2022-06-02:
We need to migrate to a new OS at some  point to get plexamp support back, while we are doing this there are several other things to consider.
- [ ] test with PlexAmp support
- [ ] investigate using pulse audio or working around it
  - [ ] check audio pipes for dynamic stream switching
  - [ ] try using some streams through docker
- [ ] better update setup

Lohrer, 2022-06-03:
Tested running:
```
sudo apt update
sudo apt upgrade
sudo apt full-upgrade
sudo rpi-update
```
Audio playback (spotify to a single zone was tested) still works after reboot.

Then upgraded to a 64-bit kernel by putting `arm_64bit=1` in `/boot/config.txt`. Stuff still works! So now there's a 64-bit kernel and 32-bit system, and everything is up-to-date.

Finally upgraded to the latest Raspberry Pi OS (from Debian Bullseye), first changed buster->bullseye in `/etc/apt/sources.list` and `/etc/apt/sources.list.d/raspi.list`. Then:
```
sudo apt update
sudo apt upgrade
```
At this point I tried to run `sudo apt full-upgrade` but ran into the error
- libc6-dev : Breaks: libgcc-8-dev (< 8.4.0-2~) but 8.3.0-6+rpi1 is to be installed

Attempting to resolve:
```
sudo apt remove libgcc-8-dev
sudo apt full-upgrade
sudo apt autoremove
sudo apt autoclean
```
At this point AmpliPi stopped working... And the config script doesn't recognize that we're on an armv7 system. I think Python is confused because it reports the platform as aarch64 even though `dpkg --print-architecture` says it's armhf. I temporarily forced the config script to accept that we are on an arm system and ran `configure.py --os-deps --python-deps`. Now everything is happy and AmpliPi works again! I suspect the switch from Python3.7->Python3.9 messed up the virtual environment.

TODO: The system is still 32-bit.

stoppablemurph, 2022-10-23:
Forgive me if I sound dumb for asking, but what exactly is left to do here? I see "The system is still 32-bit", but I'm not sure what all would need to be done to resolve this.

linknum23, 2022-10-24:
The CM3 module has a 64-bit pi on it that we are running a 32-bit OS on.

There are 2 mains tasks to do a 64-bit upgrade:
- [ ] Test the AmpliPi software and of the existing 
- [ ] Determine an upgrade path for existing system. Can it be as simple as an upgrade? Unlikely, but it is technically possible.

The complicated part if finding an upgrade path that works and isn't too cumbersome for the user..

Lohrer, 2022-10-25:
@stoppablemurph, "system is 32-bit" means all userspace packages are still 32-bit. Turns out trying to change an entire running OS from 32-bit to 64-bit on-the-fly is tricky and error-prone :laughing:

The other option for ending up with a 64-bit system would require flashing an entire pre-made image on the Pi through the service USB port, like we do at MicroNova before shipping. Not impossible, but certainly a lot more annoying than pressing a button.