micro-nova/AmpliPi #336: Testing - Setup Home Assistant at Jason's

Lohrer, 2022-06-02:


rickballard, 2022-07-05:
As AmpliPi gets increasingly productized, te help files may need to target a less technical audience.  This often means more pictures and diagrams, or even videos.  If an annotated-video is too much to ask for at the moment, please could you post a series of photos of Jason's setup process, so we can follow along?  Does he plan to host Home Assistant on his AmpliPi's rasberry Pi, or on a same-LAN connected PC or NAS?  If these are very different procedures, can you include advice on the differences?  Thanks!

linknum23, 2022-07-05:
Consumer documentation is something we are working on getting better at, we
have been lucky that most of our users are tech savvy.

As for our HomeAssistant install there is a reason that this issue exists,
it is not done yet :) At the moment we have a seperate Pi4 with a large SD
card that we are hosting HomeAssistant on for development and testing with
AmpliPi. We followed the steps here:
https://www.home-assistant.io/installation/raspberrypi

We have no plans to install HomeAssistant on AmpliPi since it logs way too
much data and will cause the embedded storage (MMC) to fail over time.

rickballard, 2022-07-05:
Definitely worth adding
https://www.home-assistant.io/installation/raspberrypi to your FAQ's page.
Thanks.  I now plan to get HA set up on my Synology NAS, and use it to add
functionality to the AmpliPi and wall-controller...

Can you consider not calling it a "wall-controller"..?  The controller is
inside the AmpliPi box, hidden away in a cabinet or rack somewhere, and the
wall-ness is not compulsory.  It can also be used like a TV-remote, or
stuck on less wall-like surfaces such as a door or lamp post.  While
"keypad" is a bit old-fashioned, it is much more likely to be a product
search term, in Google.  Or, maybe you could call it an "AmpliPiSonoff
Panel", as that name implies it is open-sourced and can include extra
open-source apps (when your software load no longer disables this feature)?

*Rick Ballard  416.414.8100 *

SteveMicroNova, 2024-10-15:
We are no longer in Jason's house, but do have HA set up in the office. Is this issue completed?

rickballard, 2024-10-15:
Yes, this is ancient history.  Thx.

closed 2024-11-04 by rtertiaer