micro-nova/AmpliPi #348: Add SiriusXM Stream

linknum23, 2022-07-06:
One of our customers has been using https://github.com/andrew0/SiriusXM in combination with our internetradio stream to add basic support for SiriusXM stations. We need to make a new Stream type to enable support for multiple streams and simplify the station selection.

vszander, 2022-07-07:
Thanks, Lincoln. 
I posted my solution (complete with instructions) in my [github repo](https://github.com/vszander/Amplipi/blob/main/python/SiriusXM.md).  If anyone is interested in participating in this stream type, feel free to follow that repository.

vszander, 2022-07-08:
I was originally under the impression that SiriusXM  would only allow __one__ streaming connection at a time / per subscription.   Usually,  if I was streaming on my computer - then switched to the SXM app on my phone,  the app would alert me with a notification that I was about to terminate the previous session if I chose to continue.   This would then be a problem with your first task - as I didn't expect that it would allow multiple simultaneous streams.

   But,  today while I left my AmpliPi streaming,   I connected and listened on my phone.  It didn't give me the warning I expected; and I returned a few hours later and the SXM stream was working and on the same channel I had left it on.    I was surprised.   
   Let's  do some testing and you can point me in a good direction for integration (especially on the other tasks ).  
 Also, anyone can signup for a 'promotional' account (  currently 3 months for $1 ) [here](https://player.siriusxm.com/freetier/welcome).   All of the cars I've purchased in the last few years have this option - and the streaming service, referenced here, is included.

   One task I would add is:   'Now Playing'   /  Metadata.    There is currently no API for this streaming service.   However, I think (after I get some integration guidance from you) -  I can adapt [this 'scraping' method](https://gist.github.com/Ionshard/1786883) .

linknum23, 2022-07-08:
Added the scraping method.

linknum23, 2023-02-27:
A more up to date fork is: https://github.com/Nintendult/SiriusXM