micro-nova/AmpliPi #357: No sound after recent update.

jonathanslenders, 2022-07-27:
After trying to update to the most recent release, I'm not getting any sound anymore. Sound works again if I downgrade to 0.1.7. Any idea what I can try to debug?

- The update always fails at the very end. I'm not sure where I can see details of what went wrong.
- On 0.1.8, sound also doesn't work when selecting an RCA input.
- Sound works again if I downgrade to 0.1.7. (I did the upgrade/downgrade path a couple of times).

I don't mind troubleshooting over SSH. 

Failure when upgrading:
[image]
Also:
[image]

Failure when downgrading:
[image]

linknum23, 2022-07-27:
Thanks for the detailed report. From what I can tell it looks like something is failing while updating the python packages during the install, since that is the next step I saw on a functional install after the shairport message.

From a debugging stance the most useful thing you could do for us would be to open an ssh connection to the pi before the 0.1.7 -> 0.1.8 update and type: ```journalctl -f``` This will print everything that is being logged in the system during the upgrade. 

To fix your issue though, I think you really just need to clear out AmpliPi's python vitrual environment. Below are the commands to do that over ssh:

```
cd ~/amplipi-dev
rm -r venv
./scripts/install_python_deps.bash # this will take about 10-15 minutes
systemctl --user restart amplipi amplipi-display amplipi-updater
```

Then try upgrading again, assuming you are on 0.1.7.
```

jonathanslenders, 2022-07-27:
Here is the journalctl output. I'll try clearing the virtualenv next.

```
Jul 20 06:18:43 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:44 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:45 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:46 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:47 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:48 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK  
Jul 20 06:18:48 amplipi sudo[1946]: jonathan : TTY=pts/1 ; PWD=/home/jonathan ; USER=root ; COMMAND=/bin/journalctl -f
Jul 20 06:18:48 amplipi sudo[1946]: pam_unix(sudo:session): session opened for user root by jonathan(uid=0)
Jul 20 06:18:49 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK         
Jul 20 06:18:50 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:52 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK        
Jul 20 06:18:52 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                                                     
Jul 20 06:18:54 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK         
Jul 20 06:18:54 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK           
Jul 20 06:18:56 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:18:56 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                                                                                                                                                    
Jul 20 06:18:58 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK                
Jul 20 06:18:58 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:00 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:00 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                        
Jul 20 06:19:01 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK              
Jul 20 06:19:02 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK  
Jul 20 06:19:02 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK                
Jul 20 06:19:04 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:04 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK                                                                                                                                   
Jul 20 06:19:05 amplipi python[814]: INFO:     10.29.0.147:53760 - "GET /update HTTP/1.1" 200 OK                                                                                                                                
Jul 20 06:19:05 amplipi python[814]: INFO:     10.29.0.147:53760 - "GET /static/js/update-ui.js HTTP/1.1" 200 OK                                                                                                                
Jul 20 06:19:05 amplipi python[814]: INFO:     10.29.0.147:53762 - "GET /static/css/jquery.dm-uploader.min.css HTTP/1.1" 304 Not Modified
Jul 20 06:19:05 amplipi python[814]: INFO:     10.29.0.147:53764 - "GET /static/css/styles.css HTTP/1.1" 304 Not Modified
Jul 20 06:19:05 amplipi python[814]: INFO:     10.29.0.147:53760 - "GET /update/version HTTP/1.1" 200 OK
Jul 20 06:19:06 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:08 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:10 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:12 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:13 amplipi python[814]: downloading update from: https://api.github.com/repos/micro-nova/AmpliPi/tarball/0.1.8
Jul 20 06:19:13 amplipi python[814]: INFO:     10.29.0.147:53762 - "POST /update/download HTTP/1.1" 200 OK
Jul 20 06:19:13 amplipi python[814]: INFO:     10.29.0.147:53762 - "GET /update/install/progress HTTP/1.1" 200 OK
Jul 20 06:19:13 amplipi python[814]: INFO:     10.29.0.147:53768 - "GET /update/install HTTP/1.1" 200 OK
Jul 20 06:19:14 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK             
Jul 20 06:19:16 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:19:18 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                 
Jul 20 06:19:20 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK                                 
Jul 20 06:19:22 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                                   
Jul 20 06:19:24 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK                      
Jul 20 06:19:26 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK                   
[... cut ...]
Jul 20 06:20:06 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:20:06 amplipi systemd[1]: apt-daily-upgrade.service: Succeeded.
Jul 20 06:20:06 amplipi systemd[1]: Started Daily apt upgrade and clean activities.
Jul 20 06:20:08 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:20:10 amplipi authbind[812]: INFO:     10.29.0.147:60416 - "GET /api HTTP/1.1" 200 OK
Jul 20 06:20:12 amplipi authbind[812]: INFO:     10.29.0.147:60418 - "GET /api HTTP/1.1" 200 OK
```

jonathanslenders, 2022-07-27:
I'm getting many of these errors when recreating the virtualenv:

```

pi@amplipi ~/amplipi-dev
 $ ./scripts/install_python_deps.bash 
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Cache entry deserialization failed, entry ignored
Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
Cache entry deserialization failed, entry ignored
Collecting pip
  Cache entry deserialization failed, entry ignored
  Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLCertVerificationError(1, '[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed: certificate is not yet valid (_ssl.c:1056)'))': /simple/pip/
```

I have ipv6.

linknum23, 2022-07-27:
Hmm it looks like the time on your amplipi is messed up. It thinks it is July 20th! Can you try installing ntp?

```
sudo apt install ntp
sudo systemctl restart ntp
```

jonathanslenders, 2022-07-27:
Thanks @linknum23 !

Everything is working great now. You can close this issue.

I guessed the date being wrong caused the upgrade to fail at some point.

linknum23, 2022-07-27:
Yup seems like it. Looks like we may need to have **ntp** installed so we don't have this problem in the future.

Added an issue for this here: #358 

Thanks for all the help debugging this!

closed 2022-07-27 by linknum23