micro-nova/AmpliPi #358: NTP needs to be installed

linknum23, 2022-07-27:
As reported in the update failure in #357 the Pi's clock can get out of sync.

Lets add installing ntp to the update process so we can make sure time is always up to date.

Lohrer, 2022-07-28:
`timedatectl` should already be installed, but I'm unsure if NTP is on by default with it.

On `amplipi13`: `timedatectl status` yields
```
timedatectl status
               Local time: Thu 2022-07-28 12:13:45 EDT
           Universal time: Thu 2022-07-28 16:13:45 UTC
                 RTC time: n/a
                Time zone: America/Detroit (EDT, -0400)
System clock synchronized: yes
              NTP service: active
          RTC in local TZ: no
```

And `apt list --installed | grep ntp` yields nothing (`ntp` is not installed).

`amplipi13` _should_ be close to an as-shipped configuration but this should be tested on a fresh install to be sure.

linknum23, 2022-08-11:
In that case we should probably just verify timedatectrl's status on update

linknum23, 2023-01-30:
Is this still an issue? We haven't had any more reports of this in 6 months. Closing for now.

closed 2023-01-30 by linknum23