micro-nova/AmpliPi #361: Unable to add streams

qscripter, 2022-08-10:
When attempting to add a new stream I'm consistently getting the following 404 error regardless of which type of stream I try to add:

```
{
  "detail": "create stream failed: max() arg is an empty sequence"
}
```

From some quick looking through the code, it looks like it might be coming from an error when building the stream object from create_stream here: https://github.com/micro-nova/AmpliPi/blob/main/amplipi/ctrl.py#L663 but I have not set up a full debug environment to verify.

linknum23, 2022-08-10:
I assume you are tracing down an issue using the AmpliPi web app?

It looks like the create stream request is incomplete. What is the JSON request being sent. You should be able to see the request in the networks tab of a Firefox/chrome web debugger.

linknum23, 2022-08-11:
Also what version of AmpliPi are you using?

qscripter, 2022-08-11:
v0.1.8

Requests generated from the Web UI or from the API docs sample requests fail. It's across every stream type, here are a couple sample POST bodies that fail with the 404:

```
{
  "logo": "https://somafm.com/img3/groovesalad-200.jpg",
  "name": "Groove Salad",
  "type": "internetradio",
  "url": "http://ice2.somafm.com/groovesalad-16-aac"
}
```

```
{
  "name": "MicroNova Spotify",
  "type": "spotify"
}
```

qscripter, 2022-08-11:
Posting a totally empty JSON file returns an error:

```
{
  "detail": [
    {
      "loc": [
        "body"
      ],
      "msg": "field required",
      "type": "value_error.missing"
    }
  ]
}
```

If I post a JSON with an empty body it creates a default config with no streams:

```
{
  "sources": [
    {
      "id": 0,
      "name": "0",
      "input": "",
      "info": {
        "name": "None",
        "state": "stopped",
        "img_url": "static/imgs/disconnected.png",
        "supported_cmds": []
      }
    },
    {
      "id": 1,
      "name": "1",
      "input": "",
      "info": {
        "name": "None",
        "state": "stopped",
        "img_url": "static/imgs/disconnected.png",
        "supported_cmds": []
      }
    },
    {
      "id": 2,
      "name": "2",
      "input": "",
      "info": {
        "name": "None",
        "state": "stopped",
        "img_url": "static/imgs/disconnected.png",
        "supported_cmds": []
      }
    },
    {
      "id": 3,
      "name": "3",
      "input": "",
      "info": {
        "name": "None",
        "state": "stopped",
        "img_url": "static/imgs/disconnected.png",
        "supported_cmds": []
      }
    }
  ],
  "zones": [
    {
      "id": 0,
      "name": "Zone 1",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    },
    {
      "id": 1,
      "name": "Zone 2",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    },
    {
      "id": 2,
      "name": "Zone 3",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    },
    {
      "id": 3,
      "name": "Zone 4",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    },
    {
      "id": 4,
      "name": "Zone 5",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    },
    {
      "id": 5,
      "name": "Zone 6",
      "source_id": 0,
      "mute": true,
      "vol": -80,
      "vol_f": 0,
      "vol_min": -80,
      "vol_max": 0,
      "disabled": false
    }
  ],
  "groups": [],
  "streams": [],
  "presets": [],
  "info": {
    "config_file": "house.json",
    "version": "0.1.8",
    "mock_ctrl": false,
    "mock_streams": false
  }
}```

qscripter, 2022-08-11:
The answer seems to be yes and if I edit the `house.json` config on the device and restart, changes I make to most parts of the file stick, but adding configurations to the `"streams": []` array don't seem to create corresponding streams in the application.

Lohrer, 2022-08-11:
Yes it is possible manually edit the config while ssh'd in. You'll need to stop the amplipi service first though. To do a full factory reset of the config:

```
systemctl --user stop amplipi
rm ~/amplipi-dev/house.json ~/amplipi-dev/house.json.bak
systemctl --user start amplipi
```

linknum23, 2022-08-11:
Looks like we need to add a button for "Factory Reset" sorry I lead you wrong on this one. I was trying to avoid ssh-ing.

qscripter, 2022-08-11:
Awesome, thanks. That did the trick. Should I close this issue or do would you like to use this to capture the bug where removing all streams prevents adding new ones?

linknum23, 2022-08-11:
Keep it open for now, we'll fix this soon.

PR #363 (2022-08-19): No streams
Fixes #361
commit 9f281933 (2022-08-19): Fix create_stream to use default stream id when no streams exist

Fixes #361
closed 2022-09-20 by linknum23