micro-nova/AmpliPi #368: interactive api test at /doc should allow selecting one command value in stead of all

marc-portier, 2022-08-22:
When one tries out the Stream 'Exec Command' endpoint (i.e. `POST /api/streams/{sid}/{cmd}`) through the available front-end testing page at [/doc](http://amplipi.local/doc) one can not select just one value from the enum - clicking any of them fills them all into the field

See screenshot:

[image]

Observation applies to firmware version 0.1.8

linknum23, 2022-08-22:
This is a bug in the api viewer tool we use, We should try using a more recent one soon to see if this is fixed.

linknum23, 2022-08-22:
This appears to be fixed in the latest release of rapidoc. We should be able to integrate a fix for this in the next release.

commit e750e265 (2022-09-07): Upgrade rapidoc

Fixes #368
PR #373 (2022-09-07): Upgrade rapidoc
Fixes #368
closed 2022-09-07 by linknum23