micro-nova/AmpliPi #381: Update to 0.19 broke AmpliPi

0lek, 2022-10-06:
Sooo... I'm a bit frustrated because I wanted to program a cool thingy for my daughter using announce, but I broke AmpliPi. And while typing the below I had the idea of downgrading. I downgraded to 0.18 and it all works now.

So essentially the issue is that after upgrading to 0.19, nothing would work. Announce would not work, spotify would not work, internet radios would not work. It seemed like it was working (AmpliPi was showing the stream playing, pyampli was showing that the announce well announced and Spotify was even showing the playing cover) - but nothing was coming out of the speakers.

I have the logs that I checked a couple of times during: https://pastebin.com/5YxazhXa

I cannot find anything there.

The other thread mentioned a MDNS issue. Could be, as my network relies on vlans and so amplipi and mobiles / computers are on different subnets... BUT: a) why would spotify show the cover? b) why would internet radio not work, when it can connect to the stream?

Any ideas? My next course of action would be to again try 0.19 and see whether it's the same.

Will keep 0.18 for now though, as it works and with 0.19 it's just a brick :/

linknum23, 2022-10-06:
This sounds a lot like the firmware got into a bad state and needed to be reset. In your case I think the downgrade fixed your problem by resetting AmpliPi. In the rare case where the firmware gets into a bad state all audio and zone controls will stop working. 

I'm assuming here that when it was in a broken state you didn't restart or unplug your amplipi? (From 0.1.9 you can actually restart and shutdown from the app/api)

If this is the case a second upgrade to 0.1.9 should solve the problem.

0lek, 2022-10-06:
I actually *did* reboot it a couple times. But that did nothing. I even rebooted it from the CLI as well. Again, nothing. I will try to upgrade once more.

linknum23, 2022-10-07:
If the update back to 0.1.9 doesn't work, this could be a problem caused by recent firmware changes.  I'm pretty sure we can have you revert the firmware while keeping the latest updates. 

BTW so sorry this issue messed up a cool announcement for your daughter. That really sucks.

0lek, 2022-10-14:
@linknum23 Didn't try 0.1.9, but the update to the LMS version did work. (0.2.0?)

linknum23, 2022-10-14:
Good to hear. I'll try to package that up into a release soon.

Still odd that we couldn't figure this out.

closed 2022-11-09 by linknum23