micro-nova/AmpliPi #384: Disabling a zone whilst playing does not stop the audio

drnichols, 2022-10-11:
Running 0.1.9 if I have a zone actively playing a source and disable it from the API/UI the zone does dissappear from the UI but the music continues to play. I expected the zone to stop playing when disabled?

Lohrer, 2022-10-11:
Yes, it would seem that as of now "disabling" a zone really only hides it from the UI and keeps it from being used in announcements. All API requests of that zone still work. We should probably mute the zone when disabling (and possibly connect the zone to NO source when #332 is implemented), disable any API requests that do not include re-enabling the zone, and fully hide the zone from the UI (it currently is still selectable from the drop-down list to move to a new source).

I imagine a disabled zone corresponding to a zone that has no speaker physically connected to it, does that match your use-case?

drnichols, 2022-10-11:
I think a typical use case is probably with no speaker connected (I have such zones currently). In the scenario I was working with today the zone did have speakers attached, I was looking to disable it (via the home assistant integration where this maps to zone media player power status) in order to remove it from the active configuration through home assistant automations. I'm playing with ways to make access to amplipi more simple and user friendly in our house with a view to making it wife and children friendly in the long term.

linknum23, 2022-12-15:
This is effectively a request to add a power button like interface to zones that keeps the rest of the zones state. Merging this with #413

closed 2022-12-15 by linknum23