micro-nova/AmpliPi #386: Zone spontaneously disabled

Juggler00, 2022-10-12:
f/w v0.1.9

I upgraded my AmpliPi to the latest f/w as soon as it came out and have now "integrated" it with Homeassistant (see my post here: https://amplipi.discourse.group/t/using-has-universal-media-player-integration/147).

Several times of the past week I have found that zone 1 (so far only this one) has been disabled. I have not done this through the WebAPI, only interacting through the media player integration in HA.

I can re-enable the zone through the WebAPI.

Juggler00, 2022-10-12:
Unless there is a background power-off, I'm not sure that is the case. I only have a single tablet interface that can access this particular zone, and I have specifically disabled the power button in both the UI and universal media player integration. If it was a background power-off, I would expect this to happen to all of the zones, instead of just this one.

Is there something I can look for in the log files for this?

Juggler00, 2022-10-12:
I've made the change. Do I need to do a Factory Default reset, or will Hardware Reset be sufficient?

linknum23, 2022-10-12:
Hardware reset, factory will wipe out your config.

Juggler00, 2022-10-13:
Thanks... done! I will report back if/when I see the disabling again.

Juggler00, 2022-10-16:
My Zone 1 spontaneously disabled twice this morning. I have looked at the logs (through the webGUI), but cannot find any reference to the string as I added to the file above. Does this indicate it is HA that is causing this?

linknum23, 2022-10-17:
No. This is still a bit of a mystery. Can you paste the modified code? I'm surprised that that message didn't fire.

Juggler00, 2022-10-17:
```
    510     try:
    511       idx, zone = utils.find(self.status.zones, zid)
    512       if idx is not None and zone is not None:
    513         name, _ = utils.updated_val(update.name, zone.name)
    514         source_id, update_source_id = utils.updated_val(update.source_id, zone.source_id)
    515         mute, update_mutes = utils.updated_val(update.mute, zone.mute)
    516         vol, update_vol = utils.updated_val(update.vol, zone.vol)
    517         vol_f, update_vol_f = utils.updated_val(update.vol_f, zone.vol_f)
    518         vol_min, update_vol_min = utils.updated_val(update.vol_min, zone.vol_min)
    519         vol_max, update_vol_max = utils.updated_val(update.vol_max, zone.vol_max)
    520         disabled, _ = utils.updated_val(update.disabled, zone.disabled)
    521         # update non hw state
    522         zone.name = name
    523         if not zone.disabled and disabled: print(f"{name}, zone {idx}, was disabled by {update.dict()}")
    524         zone.disabled = disabled
```

linknum23, 2022-10-17:
Yeah that looks correct! I'm still puzzled how this is happening.

Juggler00, 2022-10-17:
I will disable the HA AmpliPi integration and see if it continues to happen. During this time I will only control the AmpliPi through webUI.