micro-nova/AmpliPi #396: Remove the concept of sources

linknum23, 2022-10-24:
After creating a working example of #262 we started thinking of reworking the source/stream concepts in the system to handle background streams
- [ ] remove all sources
- [ ] make RCA inputs a special stream-like thing
- [ ] replace streams with the concept of an active stream, will this be the place where info moves?
- [ ] split the stream.connect/disconnect concept into activate/deactivate, connect/disconnect where connect connects a loopback device or RCA device to one of the 4 sources and activate starts a stream or background stream
- [ ] replace source_id field in groups and zones with stream_id?
- [ ] #330 
- [ ] add some dynamic event handling that would happen (in example) when a user starts playing to 'Office' Spotify device

On the UI side we will need to start looking into:
- [ ] #211 
- [ ] Active stream management cmds={restart, activate, deactivate} streams

rickballard, 2022-11-11:
This is a very good idea

linknum23, 2023-02-01:
Initial work on this is being done in https://github.com/micro-nova/AmpliPi/tree/virt-streams which allows streams to be run in the background.