micro-nova/AmpliPi #400: Echo when switching to some internet radio stations

linknum23, 2022-11-09:
A discourse user reported an echo when changing internet radio stations. This likely means that the previous radio station continues to play. 
- [ ] Need a repeatable failing transition (url10 -> (url2)
- [ ] Investigate potential fixes in the run_vlc.py script used to play internet radio stations

stamateviorel, 2022-11-10:
yes i wanted to add that from what i can see is mostly aac compressed streaming.

linknum23, 2022-11-21:
Need to check into reported failures on this station: http://icecast-servers.vrtcdn.be/klara.aac specifically with announcements,

linknum23, 2025-02-05:
This has been reported to still occur on some stations: http://icecast.vrtcdn.be/ra2vlb-high.mp3